stm32duino/Arduino_Core_STM32 #792: [Question] Is it possible to use 2x USB Host MSC (Pendrive) and HID (Keyboard) at the same time with this core?

rtek1000, 2019-11-25:
I found a project that uses HAL driver (stm32f4xx_ll_usb.c), this driver already seems to be embedded in this core, I would like to know how I can use it this way: HID Host mode FS USB port (to receive commands from a keyboard) and MSC Host mode HS USB port (to access a USB stick). I have a F407VG board. Thank you.

Example found:

main.c:
>  * Example has enabled 2 USB ports (FS and HS).
>  * On FS port, HID device is expected where you will see on LCD which type is connected.
>  * In case of HS port, USB flash key should be connected where text will be written to file!

Source:
https://github.com/MaJerle/stm32fxxx_hal_libraries/tree/master/21-STM32Fxxx_USB_HOST_HID_MSC

fpistm, 2019-11-26:
Hi @rtek1000 
Currently not.
There is no USB host implemented.
The ST USB Host library is available but not used at this time.

fpistm, 2019-11-28:
I've added USB support request in the USB features request list -->#687

closed 2019-11-28 by fpistm

rtek1000, 2022-01-17:
https://github.com/stm32duino/Arduino_Core_STM32/pull/1196