stm32duino/Arduino_Core_STM32 #798: Improvement: Enable LTO

matthijskooijman, 2019-11-27:
It would be nice if LTO (link-time-optimization) is enabled on the compiler commandline, which allows much more efficient inlining and optimization.

However, there does seem to be a bug currently related to the way interrupt vectors are defined in the STM32 HAL. In particular, if you define a custom interrupt handler in your sketch, the linker does not seem to use it and keeps using the weak alias to `Default_Handler` instead (defined in CMSIS, e.g. [for F401](https://github.com/stm32duino/Arduino_Core_STM32/blob/0b9a5919751851849e957df1027fabaa6653ce1a/system/Drivers/CMSIS/Device/ST/STM32F4xx/Source/Templates/gcc/startup_stm32f401xc.s#L125-L128).

I think this is the same problem as https://gcc.gnu.org/bugzilla/show_bug.cgi?id=83967 (which is marked as fixed, but it seems the problem is only partially fixed).

This problem does not occur for interrupt handlers defined in the core with LTO enabled, I suspect because of the ordering of files on the linker commandline (and/or the fact that the core files are linked through an .a file).

The above problem probably needs to be fixed or worked around before LTO can be enabled, but this issue can serve for discussion about this.

fpistm, 2019-11-27:
LTO can be enabled in the Optimization menu.
You are right there are several issue using it. I'm facing several issue related to the toolchain.
LTO is a nice feature but not really mature, I saw several issue raised around it in each toolchain release... this is a real "sea snake".
Related to #677

matthijskooijman, 2019-11-27:
> LTO can be enabled in the Optimization menu.

Oh, missed that, thanks.

> LTO is a nice feature but not really mature, I saw several issue raised around it in each toolchain release... this is a real "sea snake".

It actually surprise me, given that it has been enabled in the AVR core for one or two years now without much problems, and I would expect gcc for ARM to be *more* mature, since it probably sees more development.

fpistm, 2019-11-27:
> It actually surprise me, given that it has been enabled in the AVR core for one or two years now without much problems, and I would expect gcc for ARM to be _more_ mature, since it probably sees [...]

Maybe I'm wrong, this is only my feeling :wink:

fpistm, 2019-12-13:
@matthijskooijman 
I close this one as LTO is already available.

closed 2019-12-13 by fpistm

matthijskooijman, 2019-12-13:
I actually thought it might be good to have an issue for discussion about LTO-related issues perhaps? Though that can also be a closed issue :-)

fpistm, 2019-12-13:
> I actually thought it might be good to have an issue for discussion about LTO-related issues perhaps?

Well original request is to "Enable LTO" which is already available :wink:
To discuss LTO related issue, a new forum will be soon up and running and this should be a best place for this.