stm32duino/Arduino_Core_STM32 #799: USB CDC not recognized

jgromes, 2019-11-27:
Hi, I'm trying to use custom STM32L452RE board as USB virtual com port. After uploading the following sketch by SWD, the L452RE shows up as "Unknown device", instead as virtual com port.

Build configuration is the following:

[image]

```c++
void setup() {
  SerialUSB.begin(115200);
}

void loop() {
  SerialUSB.println("Hello World!");
  delay(1000);
}
```

I'm fairly sure the hardware is OK - other features of the STM32 core work. Upon connecting the USB device for the first time (before uploading anything), it was recognized as "STM32 BOOTLOADER" by the system.

Is there something I'm missing, e.g. something in the variant definition that must be changed to get USB Serial support?

Thank you very much

fpistm, 2019-11-27:
As the Nucleo-L452 was introduced before the USB and it does not have an USB port, I guess the `SystemClock_Config` does not configure the USB clock.
You can try to set it by adding your own `SystemClock_Config` at the sketch level as it is a weak.

jgromes, 2019-11-27:
Thanks for the quick reply. I added the following code to L452RE variant.cpp file (just to test it out):

```c++
RCC_PeriphCLKInitTypeDef PeriphClkInit = {};
PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_USB;
PeriphClkInit.UsbClockSelection = RCC_USBCLKSOURCE_PLLSAI1;
if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInit) != HAL_OK) {
  Error_Handler();
}
```

Good news is Windows no longer reports it as "Unknown device" - bad news is that it doesn't show up at all.

Is there some board/variant that has implemented USB support, which I could use to see what else needs to be set up?

fpistm, 2019-11-27:
You have to ensure the freq is 48MHz :wink:

jgromes, 2019-11-27:
OK, so `SystemClock_Config` is missing PLLSAI1 configuration to output 48 MHz for the USB, right?

jgromes, 2019-11-27:
Yep, that seems to have done the trick. Thanks a lot, you've been a huge help.

In case anyone comes across the same issue, here's the snippet that was added to `SystemClock_Config()`:

```c++
RCC_PeriphCLKInitTypeDef PeriphClkInit = {};
PeriphClkInit.PeriphClockSelection = RCC_PERIPHCLK_USB;
PeriphClkInit.UsbClockSelection = RCC_USBCLKSOURCE_PLLSAI1;
PeriphClkInit.PLLSAI1.PLLSAI1Source = RCC_PLLSOURCE_MSI;
PeriphClkInit.PLLSAI1.PLLSAI1M = 1;
PeriphClkInit.PLLSAI1.PLLSAI1N = 24;
PeriphClkInit.PLLSAI1.PLLSAI1P = RCC_PLLP_DIV7;
PeriphClkInit.PLLSAI1.PLLSAI1Q = RCC_PLLQ_DIV2;
PeriphClkInit.PLLSAI1.PLLSAI1R = RCC_PLLR_DIV2;
PeriphClkInit.PLLSAI1.PLLSAI1ClockOut = RCC_PLLSAI1_48M2CLK;
if (HAL_RCCEx_PeriphCLKConfig(&PeriphClkInit) != HAL_OK) {
  Error_Handler();
}
```

closed 2019-11-27 by jgromes