stm32duino/Arduino_Core_STM32 #805: GPIO pin state does not change

jgromes, 2019-11-29:
Hi,

I came across a strange issue. On STM32L452RE, the following code should toggle PC1 high and low:

```c++
#define PIN    PC1

void setup() {
  pinMode(PIN, OUTPUT);
}

void loop() {
  digitalWrite(PIN, LOW);
  delayMicroseconds(5);
  digitalWrite(PIN, HIGH);
  delayMicroseconds(5);
}
```

When checking the hardware with a scope however, PC1 state does not change (it's constantly high due to external pullup). PC1 is the only pin showing this behavior - all others work fine. I ran a similar sketch that toggles all pins (apart from those used for SWD and crystal), but still, PC1 seems to be completely inaccessible.

When not using Arduino core and only HAL libraries (compiled in Atollic TrueStudio), this does not happen, PC1 works exactly as expected, so it's most likely not a hardware issue or blown pin.

Any pointers to what could be the cause would be greatly appreciated. I tried removing all alternate functions from PC1 in PeripheralPins.c, but this seems to have had no effect.

fpistm, 2019-11-29:
Hi @jgromes 
Right, there is a typo issue in the variant.h:
https://github.com/stm32duino/Arduino_Core_STM32/blob/9d0269c204dc9db5625d5bdb44e333c10db9a9d8/variants/NUCLEO_L452RE/variant.h#L83-L85

should be 49

fpistm, 2019-11-29:
> Any pointers to what could be the cause would be greatly appreciated. I tried removing all alternate functions from PC1 in PeripheralPins.c, but this seems to have had no effect.

About that, this is normal as AF is not used for basic GPIO usage.

closed 2019-11-29 by fpistm

jgromes, 2019-11-29:
Can confirm this is working on the hardware now, thanks :)

> About that, this is normal as AF is not used for basic GPIO usage.

Yep, I probably could have noticed PC1 pin number in variant.h doesn't exactly match the others.

fpistm, 2019-11-29:
No worry :wink; and welcome.
Thanks for reporting.

commit 119d793c (2019-11-29): [NUCLEO_L452RE] Fix PC1 pin number

Fixes #805