stm32duino/Arduino_Core_STM32 #808: Changing Serial Port pins results in no output

GeorgeIoak, 2019-12-01:
- I've tried this both with the latest release and a pull of the latest github 

- I've tried this both with Arduino and VSCode with PlatformIO

I have a custom board built with the STM32F405RG and I’m trying to get some debug code sent out to UART4 (Tx=PA0, Rx=PA1). I’ve tried everything I can think of but the only thing that worked was to define a SoftwareSerial but that won’t allow me to easily debug other libraries I’m using.
I created a simple project in STM32CubeIDE and confirmed that I just need to define UART4 and with the hardware connections the same I am getting test messages.

I also connected a logic analyzer and checked all available UART Tx pins to see if there was any traffic on any Tx pin and there is not.

- I've tried just Serial1.begin(9600) on all ports
- I've tried HardwareSerial(PA1, PA0)
- I've tried Serial.setRx(PA1); Serial.setTx(PA0);

Nothing seems to work but I've confirmed that my hardware connections are correct since SoftwareSerial works as well as HAL code with the STM32CubeIDE

I've also tried this from my Desktop running Win10 as well as a laptop also running Win10.

I'm not sure where to begin to try and debug this so let me know if I need to provide any additional information

fpistm, 2019-12-02:
Hi @GeorgeIoak 
Which variant you used ?

GeorgeIoak, 2019-12-02:
@fpistm I first tried with Nucelo_F401RE and then made my own variant
[IOTech_STM32F405.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3911635/IOTech_STM32F405.zip)

fpistm, 2019-12-02:
@GeorgeIoak 
As the Serial instance you defined is 4:
`#define SERIAL_UART_INSTANCE    4`

. Then generic `Serial` is mapped to `Serial4`, so
` Serial1.begin(9600)` could not work.

If using Serial.begin(9600) does not work then I guess you have a clock config issue.
Issue you met is linked to your variant.

Moreover it seems strange your variant build as it is:
```
  PC-14,     // PC_14 OSC32_1
  PC-15,     // PC_15 OSC32_2
```

GeorgeIoak, 2019-12-02:
Yes, what I want to do is map UART4 to the standard Arduino Serial port. The reason for this is that I am trying to debug why some libraries work with the Nucleo64-F401RE board but not on my custom board which so far has worked for all standard GPIO tests. If I can map the standard Serial to UART4 then I can just check the default debug messages that are already defined in the libraries.

Why do you say that PC-14/PC-15 are strange? My board does not have an external xtal so those are mapped as standard GPIO pins.

I will check again but Serial.begin(9600) did not work. Using HAL commands does work so that is able to configure the clocks correctly so I hope that you can help determine what might be worng.

fpistm, 2019-12-02:
> Yes, what I want to do is map UART4 to the standard Arduino Serial port.

So, based on the variant you shared it is OK: `Serial` == `Serial4` and use PA1/PA0.

> I will check again but Serial.begin(9600) did not work.

I don't know why it's not work, what I can said is we have the Adafruit Feather F405 board which works fine, so my first guess is your clock config is not correct or maybe the linker script.

> Why do you say that PC-14/PC-15 are strange? My board does not have an external xtal so those are mapped as standard GPIO pins.

Simply because your use a '-' and this should be an '_', so you should have an build error like this:

```make
error: 'PC' was not declared in this scope
   PC-14,
   ^~
```

GeorgeIoak, 2019-12-02:
I see now about PC-14 vs PC_14. I did not get a compile error which is strange but I have fixed that now.

Perhaps it does have to do with the linker script or clock config. I am away from the office and do not have STM32CubeIDE installed on my laptop. Do you happen to know an example that I can copy into my variant folder? I just need, no external clock, Serial on UART 4 and SPI on SPI3

fpistm, 2019-12-02:
> I see now about PC-14 vs PC_14. I did not get a compile error which is strange but I have fixed that now

Yes very strange...

> Perhaps it does have to do with the linker script or clock config. I am away from the office and do not have STM32CubeIDE installed on my laptop. Do you happen to know an example that I can copy into my variant folder? I just need, no external clock, Serial on UART 4 and SPI on SPI3

Simply use STM32CubeMX

GeorgeIoak, 2019-12-03:
I installed STM32CubeMX and configured a project based on the STM32F405RG which resulted in the following clock configuration:

`void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {0};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {0};

  /** Configure the main internal regulator output voltage 
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
  /** Initializes the CPU, AHB and APB busses clocks 
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_NONE;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK)
  {
    Error_Handler();
  }
  /** Initializes the CPU, AHB and APB busses clocks 
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK|RCC_CLOCKTYPE_SYSCLK
                              |RCC_CLOCKTYPE_PCLK1|RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSI;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV1;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV1;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_0) != HAL_OK)
  {
    Error_Handler();
  }
}`

This was the same as what I had in variant.cpp.

I also copied the contents of STM32F405RGTX_FLASH.ld into ldscript.ld

After doing this and creating a simple serial program I still am not getting any output on PA0/PA1

fpistm, 2019-12-03:
A blink works?
Send me rhe updated variant and the boards .txt, please.

fpistm, 2019-12-03:
And which version of the core you used?

GeorgeIoak, 2019-12-03:
I had a program working with this board that set and read different GPIO pins. I am also able to use SoftwareSerial and receive data on UART4.

I am using the latest release and I have also tried cloning the current repo and using that.

[IOTech_STM32F405.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3914843/IOTech_STM32F405.zip)

I have been experimenting with changes to the variants but this is what is in the C:\Users\george\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.7.0\variants\IOTech_STM32F405 folder now

fpistm, 2019-12-03:
@GeorgeIoak 
I've made some test with the Adafruit Feather F405RG.
I could not test UART4 as pins is not connected anyway I've changed the clock config to use HSI and the USART3 works without any issue with this config:

 ```c++
void SystemClock_Config(void)
{
  RCC_OscInitTypeDef RCC_OscInitStruct = {};
  RCC_ClkInitTypeDef RCC_ClkInitStruct = {};

  /** Configure the main internal regulator output voltage
  */
  __HAL_RCC_PWR_CLK_ENABLE();
  __HAL_PWR_VOLTAGESCALING_CONFIG(PWR_REGULATOR_VOLTAGE_SCALE1);
  /** Initializes the CPU, AHB and APB busses clocks
  */
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  RCC_OscInitStruct.PLL.PLLState = RCC_PLL_ON;
  RCC_OscInitStruct.PLL.PLLSource = RCC_PLLSOURCE_HSI;
  RCC_OscInitStruct.PLL.PLLM = 8;
  RCC_OscInitStruct.PLL.PLLN = 168;
  RCC_OscInitStruct.PLL.PLLP = RCC_PLLP_DIV2;
  RCC_OscInitStruct.PLL.PLLQ = 7;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    Error_Handler();
  }
  /** Initializes the CPU, AHB and APB busses clocks
  */
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK
                                | RCC_CLOCKTYPE_PCLK1 | RCC_CLOCKTYPE_PCLK2;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_PLLCLK;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV1;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV4;
  RCC_ClkInitStruct.APB2CLKDivider = RCC_HCLK_DIV2;

  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_5) != HAL_OK) {
    Error_Handler();
  }
}
```

GeorgeIoak, 2019-12-03:
OK, this is a bit odd. I tried using this even though I see on Adafruit's design that they have a 12MHz xtal attached to PH0/PH1 and my board does not. Using this clock config I was able to immediately access UART4 using either Serial4 or since my variant.h has `#define SERIAL_UART_INSTANCE    4` just using Serial also placed the sample code output onto UART4 pins.

fpistm, 2019-12-03:
So this is ok?

GeorgeIoak, 2019-12-04:
Can you explain why using `RCC_OscInitStruct.PLL.PLLM = 8;` works with a board that does not have an external crystal. From what I read PLL_M should be set to 16.

Even though this now appears to work in Arduino copying over the same settings into PlatformIO (added to VS Code) does not work

fpistm, 2019-12-04:
The PLL source can be the HSI and allow to have the max system clock frequency: 16 MHz.
With your config you set the HSI as system clock source then the maximum frequency you can reached was 16MHz.
Maybe this is the reason why it's not work with your config if you request a too high baudrate maybe at 16MHZ it was not possible.

For PIO, I could not help as I do not support/know it.
I close this one as it's work.

closed 2019-12-04 by fpistm