stm32duino/Arduino_Core_STM32 #812: Hardware Timer Input capture not working

mogupta, 2019-12-03:
Hello Everyone
I tried the following example 
`https://github.com/stm32duino/STM32Examples/blob/master/examples/Peripherals/HardwareTimer/InputCapture/InputCapture.ino
`
My observations
1. I connected a PWM signal (from another circuit) to pin D2 but it doesn't recognize the input. 
2. If I generate a PWM signal from the STM32 itself and connect it to D2 it works no problem. 
3. If I add a very small resistor 20 ohms between the STM32 generated PWM signal to D2 , it doesn't work. Frequency is 0
4. The PWM signal from another circuit obviously has some resistance. So this could be the reason why its not being detected on the STM32 ? 
5. If I connect the PWM signal (from another circuit) to Arduino Uno, it works out of the box. No problem.

I would appreciate any help. I am new to STM32 family

ABOSTM, 2019-12-03:
Hi @mogupta ,
Can you give us more details on your setup:
Which board are you using ?
Which frequency/duty cycle is your PWM from another circuit ?
Is you PWM from another circuit 5V or 3,3V ?
Warning: Arduino Uno is basically 5V whereas STM32 is 3,3V. Even if some STM32 pin are tolerante to5V it is not the case of all pins.

Which frequency is your STM32 generated PWM ?
My 1st concern is : are you in the frequency range of detection of this example ?
There is a comment about that in the sketch:
```
  // With a PrescalerFactor = 1, the minimum frequency value to measure is : TIM counter clock / CCR MAX
  //  = (SystemCoreClock) / 65535
  // Example on Nucleo_L476RG with systemClock at 80MHz, the minimum frequency is around 1,2 khz
  // To reduce minimum frequency, it is possible to increase prescaler. But this is at a cost of precision.
  // The maximum frequency depends on processing of the interruption and thus depend on board used
  // Example on Nucleo_L476RG with systemClock at 80MHz the interruption processing is around 4,5 microseconds and thus Max frequency is around 220kHz
```

mogupta, 2019-12-03:
HI @ABOSTM 
My answers are below

Can you give us more details on your setup:
Which board are you using ? **stm32f103c8t6**
Which frequency/duty cycle is your PWM from another circuit ? **Its 50Hz to 100Hz**
Is you PWM from another circuit 5V or 3,3V ? **I have tried both 5v and 3.3v**
Warning: Arduino Uno is basically 5V whereas STM32 is 3,3V. Even if some STM32 pin are tolerante to5V it is not the case of all pins. **I am aware of this**

Which frequency is your STM32 generated PWM ? **I tried 50Hz , 100Hz  - all work**
My 1st concern is : are you in the frequency range of detection of this example ? **Since it is detecting PWM from STM32 at 50Hz, it shouldn't be a problem I guess**

**More observation.** 
1. If I attach a resistor of 20 ohms between STM32 PWM and STM32 Input pin, it doesn't work. Not sure why. In Arduino , a small resistance doesn't make a difference. 
2. I attached a oscilloscope on another circuit PWM signal, it shows the right signals but as soon as I attach the STM32 input pin, the signal goes to zero volts. As if the input signal is acting as ground.

**Code is attached**
[stm32blue.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3915586/stm32blue.zip)

ABOSTM, 2019-12-03:
Hi @mogupta ,
Several points:
* I tested on my side with the blue pile stm32f103c8t6, and found an issue linked to specific GPIO management for STM32F1xx family: for input alternate fonction, GPIO must be configured in input (and not in alternate fonction like other STM32 families).
Patch is under preparation. In the meantime, as a workaround, you can configure input mode for your pin after setMode():
` pinmode(pin, INPUT);`
* Also I maintain what I said: 
  minimum input frequency for this example is (SystemCoreClock) / 65535.
  With blue pile stm32f103c8t6 running at 72MHz, --> around 1,1 kHz.
  Below this frequency the example consider frequency = 0 (too low to be measure).
   If you need low frequency, as said in the comment, you can increase prescaler.
* With the 2 issue above, I don't know how you succeed to make it work with STM32 as generator. 
  Whatever, those 2 issues should be solved first before retesting.

mogupta, 2019-12-04:
@ABOSTM It worked. Thank you very much. Please let us know whenever the fix it out. 
you are right , I had to change the prescaler to 16 or higher to get the correct frequency. 
I think we can tag this as bug until its resolved ?

PR #824 (2019-12-09): HardwareTimer: Fix STM32F1 specific gpio configuration in case of input
**Summary**

On F1 family, input alternate function must configure GPIO in input mode
(not in alternate function like other STM32 families).

**Closing issues**

Fixes #812
commit 7b41faf8 (2019-12-09): HardwareTimer: Fix STM32F1 specific gpio configuration is case of input

On F1 family, input alternate function must configure GPIO in input mode
(not in alternate function like other STM32 families).

Fixes #812
closed 2019-12-09 by fpistm