stm32duino/Arduino_Core_STM32 #813: BUG: build_opt.h causes some sort of corruption, or cache not properly cleared

AndKe, 2019-12-03:
[TEST_.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/3916563/TEST_.zip)
**Describe the bug**

In an application with build_opt.h is built first it will fail, and another application, w/o that file will fail too.
In an application without that file is compiled first (successfully), then both will compile.

**To Reproduce**
Steps to reproduce the behavior:
-download the attached file.
-open both projects side-by-side one in each Arduino 1.8.10 (release)
(I open one arduino, then select file.. and open the next project the same way, not sure whatever these two are perfectly separated instances)
Compile TEST_no_D :  observe that it is a success.
Compile TEST_with_D :  observe that it is a success.
close both Arduino IDE's 
open Arduino, and again , open both projects.
Compile TEST_with_D :  observe that it fail
Compile TEST_no_D : observe that it fail too..

```
/home/andre/.arduino15/packages/STM32/tools/arm-none-eabi-gcc/8.2.1-1.7/bin/arm-none-eabi-gcc -mcpu=cortex-m3 -mthumb -Os --specs=nano.specs -Wl,--defsym=LD_FLASH_OFFSET=0 -Wl,--defsym=LD_MAX_SIZE=65536 -Wl,--defsym=LD_MAX_DATA_SIZE=20480 -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--entry=Reset_Handler -Wl,--unresolved-symbols=report-all -Wl,--warn-common -T/home/andre/.arduino15/packages/STM32/hardware/stm32/1.7.0/variants/PILL_F103XX/ldscript.ld -Wl,-Map,/tmp/arduino/TEST_no_D.ino.map -L/home/andre/.arduino15/packages/STM32/tools/CMSIS/5.5.1/CMSIS/DSP/Lib/GCC/ -larm_cortexM3l_math -o /tmp/arduino/TEST_no_D.ino.elf -L/tmp/arduino -Wl,--start-group /tmp/arduino/sketch/SrcWrapper.cpp.o /tmp/arduino/sketch/TEST_no_D.ino.cpp.o /tmp/arduino/libraries/SrcWrapper/syscalls.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_adc.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_adc_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_can.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_cec.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_comp.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_comp_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_cordic.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_cortex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_crc.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_crc_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_cryp.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_cryp_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_dac.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_dac_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_dcmi.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_dcmi_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_dfsdm.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_dfsdm_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_dma.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_dma2d.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_dma_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_dsi.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_eth.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_eth_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_exti.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_fdcan.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_firewall.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_flash.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_flash_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_flash_ramfunc.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_fmac.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_fmpi2c.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_fmpi2c_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_gfxmmu.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_gpio.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_gpio_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_hash.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_hash_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_hcd.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_hrtim.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_hsem.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_i2c.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_i2c_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_i2s.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_i2s_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_ipcc.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_irda.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_iwdg.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_jpeg.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_lcd.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_lptim.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_ltdc.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_ltdc_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_mdios.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_mdma.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_mmc.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_mmc_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_msp_template.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_nand.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_nor.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_opamp.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_opamp_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_ospi.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_pccard.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_pcd.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_pcd_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_pka.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_pwr.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_pwr_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_qspi.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_ramecc.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_rcc.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_rcc_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_rng.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_rtc.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_rtc_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_sai.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_sai_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_sd.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_sd_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_sdadc.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_sdram.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_smartcard.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_smartcard_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_smbus.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_spdifrx.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_spi.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_spi_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_sram.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_swpmi.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_tim.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_tim_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_timebase_rtc_alarm_template.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_timebase_rtc_wakeup_template.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_timebase_tim_template.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_tsc.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_uart.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_uart_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_usart.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_usart_ex.c.o /tmp/arduino/libraries/SrcWrapper/HAL/stm32yyxx_hal_wwdg.c.o /tmp/arduino/libraries/SrcWrapper/LL/stm32yyxx_ll_adc.c.o /tmp/arduino/libraries/SrcWrapper/LL/stm32yyxx_ll_bdma.c.o /tmp/arduino/libraries/SrcWrapper/LL/stm32yyxx_ll_comp.c.o /tmp/arduino/libraries/SrcWrapper/LL/stm32yyxx_ll_cordic.c.o /tmp/arduino/libraries/SrcWrapper/LL/stm32yyxx_ll_crc.c.o /tmp/arduino/libraries/SrcWrapper/LL/stm32yyxx_ll_crs.c.o /tmp/arduino/libraries/SrcWrapper/LL/stm32yyxx_ll_dac.c.o /tmp/arduino/libraries [... cut ...]
**Desktop (please complete the following information):**
 - OS: Linux
 - Arduino IDE version: 1.8.10
 - STM32 core version: 1.7.0 +some files from master to get half_duplex to work (and it works fine)
  Arduino_Core_STM32
  Board: Generic STMF1
  Board Part : Bluepill F103C8
  U(S)ART support: Enabled(generic Serial)
  USB Support: CDC(generic "serial" supersede U(S)ART"
  USB Speed: Low/Full Speed
  Optimize: Smallest (-Os default)
  C Runtime Library : newlib Nano(default)
  Upload Method: STM23CubeProgrammer(SWD)  (ST_LINKV2)

fpistm, 2019-12-03:
@AndKe 
First, when you do a package ensure to zip only the sketch files not the .git/ directory, this will avoid to get an 17Mb archive :wink:

The TEST_with_D is not OK as you define explicitly the `Serial2`:
`HardwareSerial Serial2(PA2);`
and ask thanks the build_opt.h to instantiate the `Serial2` so the fail are normal.

Finally about why the TEST_no_D is failed is linked to the core.a archive.
I do not know how Arduino handles this but for both sketch the same cache is used
/tmp/arduino_cache_xxxxxx/core/core_xxxxxxxxxxxxxxxxxxxxxxxxxx.a

So if the first build is TEST_with_D then it's failed then the TEST_no_D will also be failed.
So if the first build is TEST_no_D then it's passed then the TEST_with_D will also be passed .

build_opt.h is not detected by the Arduino IDE as a build item and I could not do anything to avoid that as it is only used in the gcc command line as I mentioned in the wiki:
https://github.com/stm32duino/wiki/wiki/Customize-build-options-using-build_opt.h#warning

So if you add or modify the build_opt file then reload Arduino IDE.

closed 2019-12-03 by fpistm

AndKe, 2019-12-03:
Thank you.
I thought I needed to -DENABLE_HWSERIAL2 -DENABLE_HWSERIAL3   just to be able to use Serial2 / Serial3 
Apparently this is redundant as HardwareSerial Serial2 does the same job?

fpistm, 2019-12-03:
You could define hardwareSerial with the name you want or enable a default instance using -Dxxxx then it is instantiate with the first pins found in peripheral pins.c array for the usartx selected or using the predefined pin PIN_SERIAL_RX/TXx in variant.h if any

fpistm, 2019-12-04:
@AndKe 
I've updated the wiki to be more precise:
https://github.com/stm32duino/wiki/wiki/API#hardwareserial

AndKe, 2019-12-04:
it is much better now.  :+1: