stm32duino/Arduino_Core_STM32 #817: USB Host Implementation

senceryazici, 2019-12-04:
Hi there,
I have an XSens MTi-G-710 device in my hand, and I use a NUCLEO-F429ZI board.
I basically want to connect the Xsens device to STM32 via USB Connection, and want to set the stm as host device. The Xsens device communicates with ACM, so I want to be able to receive & send serial connection bytes through USB,  just like I do with a PC on /dev/ttyACM0.

I have no clue where to start, because no example is based on stm32duino but I need to use stm32duino.

How can I implement such structure ? 
I've tried the following:

```C++
#define HAL_HCD_MODULE_ENABLED

#include "Arduino.h"
#include <usbh_lib/usbh_cdc.h>
// #include "stm32f4xx_hal.h"
// #include "stm32f4xx_hal_def.h"

typedef enum {
  APPLICATION_IDLE = 0,  
  APPLICATION_START,
  APPLICATION_READY,    
  APPLICATION_RUNNING,
  APPLICATION_DISCONNECT,
} CDC_ApplicationTypeDef;

class XSensUSBDriver
{
private:
    CDC_ApplicationTypeDef Appli_state = APPLICATION_IDLE;
    USBH_HandleTypeDef hUSBHost;
    static CDC_LineCodingTypeDef LineCoding;
    static CDC_LineCodingTypeDef DefaultLineCoding;
    void receive();

public:
    // void USBH_UserProcess(USBH_HandleTypeDef *phost, uint8_t id);
    void process();
    void init();
    XSensUSBDriver();
    ~XSensUSBDriver();
};
void USBH_UserProcess(USBH_HandleTypeDef *phost, uint8_t id)
{
  switch(id)
  { 
  case HOST_USER_SELECT_CONFIGURATION:
    break;
    
  case HOST_USER_DISCONNECTION:
    // Appli_state = APPLICATION_DISCONNECT;
    break;
    
  case HOST_USER_CLASS_ACTIVE:
    // Appli_state = APPLICATION_READY;
    break;
    
  case HOST_USER_CONNECTION:
    // Appli_state = APPLICATION_START;
    break;

  default:
    break; 
  }
}

XSensUSBDriver::XSensUSBDriver()
{
    
}

void XSensUSBDriver::init()
{
    USBH_Init(&hUSBHost, USBH_UserProcess, 0);
    // USBH_RegisterClass(&hUSBHost, USBH_CDC_CLASS);
    

    // ?
    // USBH_CDC_GetLineCoding(&hUSBHost, &DefaultLineCoding);
    // LineCoding = DefaultLineCoding;
    // LineCoding.b.dwDTERate = 115200; // baudrate
    // LineCoding.b.bParityType = 0; // 0, 1, 2
    // LineCoding.b.bDataBits = 8; // data bits
    // LineCoding.b.bCharFormat = 1; // stop bits
    // USBH_CDC_SetLineCoding(&hUSBHost, &LineCoding);
}

void XSensUSBDriver::process()
{
    USBH_Process(&hUSBHost); 
}

void XSensUSBDriver::receive()
{
    uint8_t recv[16];
    USBH_CDC_Receive(&hUSBHost, recv, 16);
}

XSensUSBDriver::~XSensUSBDriver()
{
}

```

And my folder structure looks like this:

├── include
│   └── README
├── lib
│   ├── README
│   └── XSensUSBDriver
│    |    ├── usbh_lib
│    |    │   ├── usb_conf.c
│    |    │   ├── usbh_cdc.c
│    |    │   ├── usbh_cdc.h
│    |    │   ├── usbh_conf.h
│    |    │   ├── usbh_conf_template.c
│    |    │   ├── usbh_conf_template.h
│    |    │   ├── usbh_core.c
│    |    │   ├── usbh_core.h
│    |    │   ├── usbh_ctlreq.c
│    |    │   ├── usbh_ctlreq.h
│    |    │   ├── usbh_def.h
│    |    │   ├── usbh_ioreq.c
│    |    │   ├── usbh_ioreq.h
│    |    │   ├── usbh_pipes.c
│    |    │   └── usbh_pipes.h
│    |    ├── XSensUSBDriver.cpp
│    |    └── XSensUSBDriver.h
├── platformio.ini
├── src
│   └── main.cpp
└── test
    └── README

Any help would be appriciated.

Sorry for my english though

fpistm, 2019-12-05:
Hi @senceryazici 
First you will have to add the USB Host library in the path like for the Device one:
https://github.com/stm32duino/Arduino_Core_STM32/blob/ad09a48db2d4c6a05491f9be91ef095580fca5f6/platform.txt#L12

Then to define `HAL_HCD_MODULE_ENABLED` you have to use the `hal_conf_extra.h`:
[HAL-configuration](https://github.com/stm32duino/wiki/wiki/HAL-configuration#core-version--150-1)

Then, you will have to wrap source files from USB Host library like it is for Device one:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/stm32/usb/usb_device_core.c

Then add all files required for USB host (like it has been done for USB device).

fpistm, 2019-12-05:
> And also I've checked #687 , are there any estimations when the USB Host library will be supported on stm32duino core ?

Currently not. Time is always missing ingredients.
Any contributions are welcome.

I close this one as now added to #687 list

closed 2019-12-05 by fpistm