stm32duino/Arduino_Core_STM32 #820: Upload fails with "NODE G431KB not found". When Compiling to Nucleo-32 STM32G431 in Arduino 1.8.10

niels4911, 2019-12-06:
It looks like the Upload fails because my board shows up as E\:NOD_G431KB in the file manager. 

When I rename the mass storage to NODE_G431KB it works - but I have to rename the E: drive every time I what to upload a new version.

fpistm, 2019-12-06:
Hi @niels4911 
Could you try to update the STLink firmware thanks STM32CubeProgrammer.

fpistm, 2019-12-06:
I've made a test with a newer board and reproduce.
It seems the most recent STLink firmware decide to change the name....
With V3J2M1: _"NODE_G431KB"_
wtih newer: _"NOD_G431KB"_

I will do an update to the boards.txt:
https://github.com/stm32duino/Arduino_Core_STM32/blob/ad09a48db2d4c6a05491f9be91ef095580fca5f6/boards.txt#L286

```patch
- Nucleo_64.menu.pnum.NUCLEO_G431RB.node=NODE_G431RB
+ Nucleo_64.menu.pnum.NUCLEO_G431RB.node="NODE_G431RB,NOD_G431RB"
```

closed 2019-12-06 by fpistm

Nivek92, 2023-12-15:
I receive a similar issue (NOD_C031C6 not found.) when trying to use it with an Nucleo-64 STM32F302R8

fpistm, 2023-12-15:
> I receive a similar issue (NOD_C031C6 not found.) when trying to use it with an Nucleo-64 STM32F302R8

Not the same, seems you didn't select the correct board, the C031 is the first one of the list....

commit 9b224d61 (2019-12-06): [Nucleo-G431xx] Add new node name

Fix #820