stm32duino/Arduino_Core_STM32 #821: DMA Support - IRQ Handlers in Core?

xC0000005, 2019-12-07:
While trying to get some LEDs working with the ST Arduino core, I ran into a problem that's likely a missing feature. Libraries probably shouldn't be implementing the weak IRQ handlers, because you can't have them defined anywhere. In my own fork, I added, similar to timers, handlers with registration/deregistration. For things like ADC and SD Support, this would be handy, but I'm not sure if there's a plan for this. I could create a PR, but I'd rather see what the consensus is.

Link to a branch with some test DMA support (ignore the library - I'm not sure how I *want* it to work, so it's not even remotely usable).

https://github.com/xC0000005/Arduino_Core_STM32/tree/DMASupport

fpistm, 2019-12-09:
Hi @xC0000005 
DMA support  would be fine.
Currently, I have no plan about that. Any contribution are welcome.
Do not hesitate to proposal a PR (you can set it as Draft) to open discussion about your proposal.

4ndreas, 2019-12-09:
I had the same problem, that you can not add IRQ functions in the arduino code (Also for DMA). My work around was to add a new board variant with the IRW functions. This is not perfect but at least I don't have to fork the hole core. 
Adding IRW functions to the variant.c works for me.

fpistm, 2019-12-09:
Hi @4ndreas 

> I had the same problem, that you can not add IRQ functions in the arduino code (Also for DMA).

I guess you talk about adding an IRD handler in the sketch ? Did you define it with extern "c" ?

4ndreas, 2019-12-09:
No In my case I needed the 

```
 void DMA2_Stream2_IRQHandler(void)
 {
   /* USER CODE BEGIN DMA2_Stream2_IRQn 0 */

   /* USER CODE END DMA2_Stream2_IRQn 0 */
   HAL_DMA_IRQHandler(&hdma_tim8_ch1);
   /* USER CODE BEGIN DMA2_Stream2_IRQn 1 */

   /* USER CODE END DMA2_Stream2_IRQn 1 */
 }
```
Function, adding this in the *.ino files won't work, but in the variant.c does. Took me a while to figure out why the interrupt jumped to a wired position.

fpistm, 2019-12-09:
At sketch level it should be:
` extern "C" void DMA2_Stream2_IRQHandler(void)`

Note:
In your case, I guess a library would be a best choice.
the Adafruit Neopixel library is now compatible with the STM32 core.

4ndreas, 2019-12-09:
> At sketch level it should be:
> ` extern "C" void DMA2_Stream2_IRQHandler(void)`
> 
> Note:
> In your case, I guess a library would be a best choice.
> the Adafruit Neopixel library is now compatible with the STM32 core.

That is something for the Wiki I gues, thank you. 
My driver drives 16 strips up top 1000 leds in parallel. The DMA controller is perfect for this.

xC0000005, 2019-12-09:
I'll open a PR this week. The goal is simple - have sketches not defining handlers for events that are global to the core, so libraries can register for different DMA events without having conflicting handlers.  Thanks for the input, all.

xC0000005, 2019-12-10:
Opened #825 since a suggested change is better than a dozen posts about what we could do.

closed 2019-12-10 by xC0000005