stm32duino/Arduino_Core_STM32 #823: Question: RAM data area that survives over system restart

lyusupov, 2019-12-08:
This is a kind of basic question that I can not find an answer on... It probably should be asked on stm32duino forum, but it seems to be down right now.

Does the Arduino/STM32 has an equivalent of RTC_DATA_ATTR attribute ? 
This attribute is defined in Arduino/ESP32 and gives an opportunity to have some data in a RAM section that does not get overwritten in a subsequent deep sleep, shutdown and restart operations.

They implement the area by means of ld script section, like this:

#define RTC_DATA_ATTR __attribute__((section(".rtc.data")))

```
  /* RTC data section holds RTC wake stub
     data/rodata, including from any source file
     named rtc_wake_stub*.c and the data marked with
     RTC_DATA_ATTR, RTC_RODATA_ATTR attributes.
     The memory location of the data is dependent on 
     CONFIG_ESP32_RTCDATA_IN_FAST_MEM option.
  */
  .rtc.data :
  {
    _rtc_data_start = ABSOLUTE(.);
    *(.rtc.data)
    *(.rtc.rodata)
    *rtc_wake_stub*.*(.data .rodata .data.* .rodata.* .bss .bss.*)
    _rtc_data_end = ABSOLUTE(.);
  } > rtc_data_location
```
Full content: 
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/ld/esp32.common.ld

Attributes:
https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/esp32/esp_attr.h

When I look into ld scripts of Arduino/STM32 I don't see anything similar.

Is the functionality already implemented in Arduino/STM32 but with some different way ?

fpistm, 2019-12-09:
Hi @lyusupov 
no there is no kind of mechanism like that with linker script.
To save some data the backup register can be used but I guess it is limited (size) vs ESP32 implementation.

lyusupov, 2019-12-09:
@fpistm 
Thank you for quick answer and very good tip!

It looks to me that L0 has at least five 32-bit backup registers.
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/Drivers/STM32L0xx_HAL_Driver/Inc/stm32l0xx_hal_rtc_ex.h

 It is likely sufficient for most typical use cases, including my one.

closed 2019-12-09 by lyusupov

lyusupov, 2020-03-19:
Seems like some kind of activity related to this topic is currently happening there: https://github.com/stm32duino/Arduino_Core_STM32/pull/996