stm32duino/Arduino_Core_STM32 #827: Serial hangs on STM32

philbowles, 2019-12-10:
## Problem:

Board hangs when Serial.begin is called from constructor of static global class. Same code works as expected on ESP8266.

## Steps to reproduce:
Run the following code:

```cpp
class hang{
  public:
    hang(){    
        Serial.begin(115200);
        Serial.print("ANYTHING"); // hangs on STM32, not on ESP8266
    }
};
hang stm32only;

void setup() {
  Serial.println("You will never see this on STM32, always on ESP8266");
}
void loop() {}
```

**Expected behavior**
Code should print "You will never see this on STM32, always on ESP8266" to Serial monitor

**Desktop (please complete the following information):**
 - OS: Windows 7
 - Arduino IDE version: 1.8.10
 - STM32 core version: 1.7.0
 - Tools menu setting: All defaults
 - Upload method: Mass Storage

**Board (please complete the following information):**
 - Name: Nucleo F429ZI 
 - Hardware Revision: MB1137 Rev B
 - Extra hardware used if any: none

**Additional context**
 
I am sure that this is related the "undefined" order of static global initialisation in different compilation units. I suspect Serial is not yet instantiated at the time of the constructor call in STM32 core, whereas it most definitely is in ESP8266.

IF Serial cannot be called in STM32 core until "setup" then this needs to be at least documented, but preferably fixed to make its behaviour the same (as expected) in the other Arduino cores.

senceryazici, 2019-12-11:
Hi Phil,

The Serial should work when the class “hang” is initialized in setup function, I’m not an expert but just to give you a quick response (until an expert replies) you may want to check the main file, where the “int main” is constructed. Because in the main the setup is called and after that the loop function is called in a loop. Before the setup function is called, possibly there should be some hardware configurations specific for your board. When the class is initialized before those configurations, possibly will result in a situation where your Serial “doesn’t work”. I’m not sure which configuration might cause the issue (maybe the clock configuration ?)

But like I said, definitely wait for someone to give you a better response, I just wanted to give you an idea where to search the cause of the problem.

fpistm, 2019-12-11:
Hi @philbowles 
Right, this is 

>  related the "undefined" order of static global initialisation in different compilation units.

> IF Serial cannot be called in STM32 core until "setup" then this needs to be at least documented, but preferably fixed to make its behaviour the same (as expected) in the other Arduino cores.

This behavior do not work for all Arduino Core. Tested with Arduino Due and this not works too.

Generally speaking this is not a good idea to call other global object in a constructor anyway do not hesitate to provide a PR.

philbowles, 2019-12-11:
Thanks everyone.
1) I know "this is not a good idea to call other global object", I wanted to know WHY it works on ESP8266 core and not on STM32 (For information, I was trying to debug the constructor - I wouldn't use this in production code!)
2) "This behavior do not work for all Arduino Core" - Yes, it will not work on cores where Serial is not fully constructed before the user sketch classes are... I wanted conformation that this was the case in STM32 core, so that a) I could avoid it b) other folk were also aware
3) I wanted to confirm it was actually the behaviour of STM32 core and not something else I was doing - this seems to be the case.
4) No PR required, this should be solved simply by stating in the documentation that Serial should not be called until Setup starts.

philbowles, 2019-12-11:
Let's just put a warning in the documentation.

closed 2019-12-11 by philbowles