stm32duino/Arduino_Core_STM32 #832: analogRead() always returns 0 when slowing down clock

urish, 2019-12-14:
**Describe the bug**
`analogRead()` always returns 0 for floating pin when running with HSI, no PLL, and sysclk divider of 4 (or greater).

I'm using a floating analog pin in my design to prime the random number generator. I noticed that whenever I slow down the clock, the RNG always returns the same numbers. When debugging, I see that `analogRead()` always returns 0 once I set the clock divider to 4 or higher.

**To Reproduce**

```arduino
void SystemClock_Config()
{
  RCC_OscInitTypeDef RCC_OscInitStruct;
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    _Error_Handler(__FILE__,  __LINE__);
  }

  RCC_ClkInitTypeDef RCC_ClkInitStruct;
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK
                                | RCC_CLOCKTYPE_PCLK1;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSI;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV4;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV16;
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_1) != HAL_OK) {
    _Error_Handler(__FILE__, __LINE__);
  }
}

setup() {
  void setup() {
  pinMode(A6, INPUT);
  int value = analogRead(A6);
  Serial.begin(115200);
  Serial.println(value);
}

void loop() {
}
```

Steps to reproduce the behavior:
1. Flash the above code, make sure A6 is floating
2. Start the program and check its output. Run it several times. I will always spit "0"
3. Change the AHCB Clock Divider to a smaller value, e.g.:
    `RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV2;`
4. Flash the program again
5. Start the program and check its output. Run it several times. You will see different numbers each time.

**Expected behavior**
I'd expect analogRead() to behave the same, regardless of the clock speed setting.

**Desktop (please complete the following information):**
 - OS: Windows
 - Arduino IDE version: 1.8.10
 - STM32 core version: 1.7.0
 - Upload method: STLink (through STM32CubeProgrammer)

**Board**
- Name: Custom board using STM32F030F4P6

urish, 2019-12-14:
Note: as a workaround, I read the analog value just before setting the clock speed:
```arduino
int value = 0;

void SystemClock_Config()
{
  value = analogRead(A6);
  RCC_OscInitTypeDef RCC_OscInitStruct;
  RCC_OscInitStruct.OscillatorType = RCC_OSCILLATORTYPE_HSI;
  RCC_OscInitStruct.HSIState = RCC_HSI_ON;
  RCC_OscInitStruct.HSICalibrationValue = RCC_HSICALIBRATION_DEFAULT;
  if (HAL_RCC_OscConfig(&RCC_OscInitStruct) != HAL_OK) {
    _Error_Handler(__FILE__,  __LINE__);
  }

  RCC_ClkInitTypeDef RCC_ClkInitStruct;
  RCC_ClkInitStruct.ClockType = RCC_CLOCKTYPE_HCLK | RCC_CLOCKTYPE_SYSCLK
                                | RCC_CLOCKTYPE_PCLK1;
  RCC_ClkInitStruct.SYSCLKSource = RCC_SYSCLKSOURCE_HSI;
  RCC_ClkInitStruct.AHBCLKDivider = RCC_SYSCLK_DIV4;
  RCC_ClkInitStruct.APB1CLKDivider = RCC_HCLK_DIV16;
  if (HAL_RCC_ClockConfig(&RCC_ClkInitStruct, FLASH_LATENCY_1) != HAL_OK) {
    _Error_Handler(__FILE__, __LINE__);
  }
}

setup() {
  Serial.begin(115200);
  Serial.println(value);
}

void loop() {
}
```

This does the trick for me, since in my case I'm only interested in reading once (to seed the RNG)

fpistm, 2019-12-14:
Probably, HAL return an error , that's why you get 0 in this case you should see why there is an error. I guess ADC run too slow.

urish, 2019-12-15:
Good spot, seems like `HAL_ADCEx_Calibration_Start()` returns `HAL_ERROR`. By looking at [the source code](https://github.com/stm32duino/Arduino_Core_STM32/blob/4b515fb5fc2dbeb2f46e0ccada3373978b000435/system/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_adc_ex.c#L144), I suspect it's timing out when calibrating the ADC. I will keep investigating.

urish, 2019-12-15:
Seems like I found [the culprit](https://github.com/stm32duino/Arduino_Core_STM32/blob/4b515fb5fc2dbeb2f46e0ccada3373978b000435/system/Drivers/STM32F0xx_HAL_Driver/Src/stm32f0xx_hal_adc_ex.c#L61). The timeout there is too low when using the above clock configuration (despite the fact the comment above that line says it accounts for the worst case). Here are the lowest timeout values that worked for me for each value of `AHBCLKDivider`:

| AHBCLKDivider | Minimum timeout |
|-----------------|----------------|
| RCC_SYSCLK_DIV4 | 4 |
| RCC_SYSCLK_DIV8 | 7 |
| RCC_SYSCLK_DIV16 | 14 |
| RCC_SYSCLK_DIV64 | 56 |

For `RCC_SYSCLK_DIV16` and above, changing the timeout alone is not enough, as there is another failure later in `HAL_ADC_Start()`.

My suggestion is to increase the timeout value to `7`, and get the first two cases (clock divider of 4 or 8) to work. Should I send a PR for this?

fpistm, 2019-12-15:
> My suggestion is to increase the timeout value to `7`, and get the first two cases (clock divider of 4 or 8) to work. Should I send a PR for this?

No, HAL are not modified.
I guess the value is correct but there is an other issue. Maybe the HSI is not ready or any other issue.

urish, 2019-12-15:
I did another test where I called `analogRead()` multiple times with different delays between, but it still fails with the default timeout, so it's unlikely an issue with HSI not ready (unless I didn't understood your suggestion).

Googling shows that they had a similar bug in other core: https://community.st.com/s/question/0D50X00009kLCgU/bug-in-haladcexcalibrationstart

fpistm, 2019-12-15:
Looking at the Product Specifications of the stm32f030f4: https://www.st.com/resource/en/datasheet/stm32f030f4.pdf

[image]

Your clock config apply a frequency of 0.125MHz which is too slow for the ADC.

You should try to use the ASYNC clock to achieve this as by default this is the PCLK.
https://github.com/stm32duino/Arduino_Core_STM32/blob/4b515fb5fc2dbeb2f46e0ccada3373978b000435/libraries/SrcWrapper/src/stm32/analog.cpp#L102-L110
To do this try to redefine the `ADC_CLOCK_DIV`  to `ADC_CLOCK_ASYNC_DIV1` in your variant.h

urish, 2019-12-15:
Brilliant! Adding `-DADC_CLOCK_DIV=ADC_CLOCK_ASYNC_DIV1` to my `built_opt.h` did the trick. I can now use `analogRead()` with `RCC_SYSCLK_DIV4`. I tested with larger dividers as well: 8, 16, and 64, and they all seem to be working well with this configuration.  Thanks!

closed 2019-12-15 by urish

fpistm, 2019-12-15:
Welcome