stm32duino/Arduino_Core_STM32 #833: Shouldn't it be possible to use different board but with MCU?

odo2063, 2019-12-14:
Hi!

I try to build a script for the Nucleo-F072RB, which is not supported.

I tried using the discovery-F072RB but it did not work.

I only need serial over ST-Link and I²C. I copied the .bin by hand to the device.

Shouldn't that work or what am I missing?

fpistm, 2019-12-14:
The serial is probably not the same.

fpistm, 2019-12-15:
Disco F072 define Serial to use USART1 with PA10/9
Nucleo F072 have to define Serial to use USART2 with PA3/2

odo2063, 2019-12-15:
how do i change that??

fpistm, 2019-12-16:
@odo2063 
they are several way to do this.
The simple one is to modify the variant.h of the DISCO_F072RB to change the related Serial definition.
You need to define `TSC_REMOVED` else the PA2 and PA3 will not be available.
```patch 
+ #define TSC_REMOVED
// UART Definitions
- #define SERIAL_UART_INSTANCE    1 // USART 1
+ #define SERIAL_UART_INSTANCE    2 // USART 2

// Mandatory for Firmata
- #define PIN_SERIAL_RX           PA10
- #define PIN_SERIAL_TX           PA9
+ #define PIN_SERIAL_RX           PA3
+ #define PIN_SERIAL_TX           PA2
```

Another way is to simply connect the PA9 and PA10 to the RX and Tx of the STLink part of the Nucleo F072RB. Then no change is needed in the core file.

I've tested and it works with a NUCLEO F072RB. I close this issue as OP ask if it is possible. Answer is yes but up to you to ensure correct pins mapping and definition.

closed 2019-12-16 by fpistm

odo2063, 2019-12-21:
ok, now i added a new variant Folder `NUCLEO_F072RB` and copied the Pinmap from F030R8 (for correct arduino mapping, but kept your changes to Serial part) and added it to `boards.txt`.

Compiling and Uploading works. but the code does not  seem to run... :-/

fpistm, 2019-12-21:
@odo2063 
It works, don't know what you do. Try to understand what you do and what is required.

odo2063, 2019-12-21:
ok...works with the changes to the Disco variant.h......but it seems i am unable(too stupid) to provide a new variant m-(

odo2063, 2019-12-21:
that's my NUCLEO_F072RBs variant.h

```
/*
 *******************************************************************************
 * Copyright (c) 2018, STMicroelectronics
 * All rights reserved.
 *
 * Redistribution and use in source and binary forms, with or without
 * modification, are permitted provided that the following conditions are met:
 *
 * 1. Redistributions of source code must retain the above copyright notice,
 *    this list of conditions and the following disclaimer.
 * 2. Redistributions in binary form must reproduce the above copyright notice,
 *    this list of conditions and the following disclaimer in the documentation
 *    and/or other materials provided with the distribution.
 * 3. Neither the name of STMicroelectronics nor the names of its contributors
 *    may be used to endorse or promote products derived from this software
 *    without specific prior written permission.
 *
 * THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS"
 * AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE
 * IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE
 * DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE
 * FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL
 * DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR
 * SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER
 * CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY,
 * OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE
 * OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
 *******************************************************************************
 */

#ifndef _VARIANT_ARDUINO_STM32_
#define _VARIANT_ARDUINO_STM32_

#ifdef __cplusplus
extern "C" {
#endif // __cplusplus

/*----------------------------------------------------------------------------
 *        Pins
 *----------------------------------------------------------------------------*/

#define PA3  0
#define PA2  1
#define PA10 2
#define PB3  3  // no PWM
#define PB5  4
#define PB4  5
#define PB10 6  // no PWM
#define PA8  7
#define PA9  8
#define PC7  9
#define PB6  10
#define PA7  11 // A6
#define PA6  12 // A7
#define PA5  13
#define PB9  14
#define PB8  15
// ST Morpho
// CN7 Left Side
#define PC10 16
#define PC12 17
#define PF6  18
#define PF7  19
#define PA13 20 // SWD
#define PA14 21 // SWD
#define PA15 22
#define PB7  23
#define PC13 24
#define PC14 25
#define PC15 26
#define PF0  27
#define PF1  28
#define PC2  29 // A8
#define PC3  30 // A9
// CN7 Right Side
#define PC11 31
#define PD2  32
// CN10 Left Side
#define PC9  33
// CN10 Right side
#define PC8  34
#define PC6  35
#define PC5  36 // A10
#define PA12 37
#define PA11 38
#define PB12 39
#define PB11 40
#define PB2  41
#define PB1  42
#define PB15 43
#define PB14 44
#define PB13 45
#define PC4  46 // A11
#define PF5  47
#define PF4  48
#define PA0  49 // A0
#define PA1  50 // A1
#define PA4  51 // A2
#define PB0  52 // A3
#define PC1  53 // A4
#define PC0  54 // A5

// This must be a literal
#define NUM_DIGITAL_PINS        61
// This must be a literal with a value less than or equal to to MAX_ANALOG_INPUTS
#define NUM_ANALOG_INPUTS       12
#define NUM_ANALOG_FIRST        49

// On-board LED pin number
#define LED_BUILTIN             13
#define LED_GREEN               LED_BUILTIN

// On-board user button
#define USER_BTN                PC13

// Timer Definitions
//Do not use timer used by PWM pins when possible. See PinMap_PWM.
#define TIMER_TONE              TIM6

//Do not use basic timer: OC is required
#define TIMER_SERVO             TIM3  //TODO: advanced-control timers don't work

// UART Definitions
#define TSC_REMOVED //You need to define TSC_REMOVED else the PA2 and PA3 will not be available.
//#define SERIAL_UART_INSTANCE    1 // USART 1
#define SERIAL_UART_INSTANCE    2 // USART 2

// Mandatory for Firmata
//#define PIN_SERIAL_RX           PA10
//#define PIN_SERIAL_TX           PA9
#define PIN_SERIAL_RX           PA3
#define PIN_SERIAL_TX           PA2

#ifdef __cplusplus
} // extern "C"
#endif
/*----------------------------------------------------------------------------
 *        Arduino objects - C++ only
 *----------------------------------------------------------------------------*/

#ifdef __cplusplus
// These serial port names are intended to allow libraries and architecture-neutral
// sketches to automatically default to the correct port name for a particular type
// of use.  For example, a GPS module would normally connect to SERIAL_PORT_HARDWARE_OPEN,
// the first hardware serial port whose RX/TX pins are not dedicated to another use.
//
// SERIAL_PORT_MONITOR        Port which normally prints to the Arduino Serial Monitor
//
// SERIAL_PORT_USBVIRTUAL     Port which is USB virtual serial
//
// SERIAL_PORT_LINUXBRIDGE    Port which connects to a Linux system via Bridge library
//
// SERIAL_PORT_HARDWARE       Hardware serial port, physical RX & TX pins.
//
// SERIAL_PORT_HARDWARE_OPEN  Hardware serial ports which are open for use.  Their RX & TX
//                            pins are NOT connected to anything by default.
#define SERIAL_PORT_MONITOR     Serial
#define SERIAL_PORT_HARDWARE    Serial
#endif

#endif /* _VARIANT_ARDUINO_STM32_ */
```