stm32duino/Arduino_Core_STM32 #834: Nucleo-G431KB, delayMicroseconds is incorrect

niels4911, 2019-12-16:
Im trying to output a 1KHz sqaure wave on the DAC output pin "PA4", however i'm only able to get 250 Hz.

My test program looks like this

void setup() {
  // put your setup code here, to run once:
   pinMode(PA4, OUTPUT); // DAC OUT
}

void loop() {
  // put your main code here, to run repeatedly:
  analogWrite(PA4, 4095 );
  delayMicroseconds(500);
  analogWrite(PA4, 0);
  delayMicroseconds(500);
}

fpistm, 2019-12-16:
Hi @niels4911 

`delayMicroseconds` is correct.

The main issue is how you want achieve this:
`analogWrite()` takes some times before really executing the DAC output, that's why you didn't get the correct freq.

Try this one:
```C++
void setup() {
  // put your setup code here, to run once:
  pinMode(PA4, OUTPUT); // DAC OUT
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWriteFast(PA_4, HIGH);
  delayMicroseconds(500);
  digitalWriteFast(PA_4, LOW);
  delayMicroseconds(500);
}
```

niels4911, 2019-12-16:
Thanks for the quick response. 

You're correct. With your code I can output at 1KHz. Thanks!!  

My goal is to output a sqaure wave or sine wave with different amplitudes. Do known of a way to get the analogWrite to output at 1KHz?

fpistm, 2019-12-16:
https://github.com/stm32duino/STM32Examples/blob/master/examples/Peripherals/HardwareTimer/All-in-one_setPWM/All-in-one_setPWM.ino

niels4911, 2019-12-17:
I'm not sure this helps with my specific goal ... but many thanks for the effort

closed 2019-12-17 by niels4911