stm32duino/Arduino_Core_STM32 #836: How to check the operating clock speed?

xtream123, 2019-12-17:
Hi, 
How we can check the actual(or the set) operating CPU speed of the mcu via code function or in .h/.cpp file/board info? 

I am using the Generic STM32F4xx (Black F407VET6) board and I want to maximize it operating CPU speed to 168MHz.

Thank you.

fpistm, 2019-12-17:
Hi @xtream123 
The current CPU Freq is avaible thanks F_CPU:
https://github.com/stm32duino/Arduino_Core_STM32/blob/8110ed411be152abdfce4ce8dae1f090c2e3359d/cores/arduino/stm32/stm32_def.h#L58

HAL RCC ensures the `SystemCoreClock` always updated to the current frequency.

The clock config is available in the variant.cpp. Here is the clock config for the Black F407:
https://github.com/stm32duino/Arduino_Core_STM32/blob/d2ebcd532164a2ed2f74f9337bd86fab766ea133/variants/BLACK_F407XX/variant.cpp#L187

You can redefine it as it is a weak function.

fpistm, 2019-12-17:
Note that I think it should already running at max speed.

xtream123, 2019-12-18:
Thank you @fpistm . I will take test it now and let you know my result.

xtream123, 2019-12-18:
I have checked the file and it looks good, I counter check the clock config with STM32CubeMX it seems okay.

I also test the F_CPU using :
 Serial.println(F_CPU);

Output:
168000000

and it seems okay.

Thank you.

fpistm, 2019-12-18:
welcome

closed 2019-12-18 by fpistm