stm32duino/Arduino_Core_STM32 #842: Firmata fails to compile on F401RE

morganrallen, 2019-12-22:
StandardFirmata fails to compile as described in wiki.

```
Arduino: 1.8.9 (Linux), Board: "Nucleo-64, Nucleo F401RE, Mass Storage, Enabled (generic 'Serial'), None, Low/Full Speed, Smallest (-Os default), Newlib Nano (default)"
********
In file included from /home/morgan/.bin/arduino-1.8.9/libraries/Firmata/examples/StandardFirmata/StandardFirmata.ino:28:
/home/morgan/.bin/arduino-1.8.9/libraries/Firmata/Firmata.h:131:17: error: friend declaration of 'void encodeByteStream(size_t, uint8_t*, size_t) const' specifies default arguments and isn't a definition [-fpermissive]
  131 |     friend void FirmataMarshaller::encodeByteStream (size_t bytec, uint8_t * bytev, size_t max_bytes = 0) const;
      |                 ^~~~~~~~~~~~~~~~~
```
 - OS: Ubuntu
 - Arduino IDE version: 1.8.9 & 1.8.10
 - STM32 core version: 1.8.0
 - Tools menu settings if not the default: default but also fails with Newlib Standard
 - Upload method: Mass Storage

 - Name: Nucleo F401RE]
 - Hardware Revision: C
 - Extra hardware used if any: None

fpistm, 2019-12-22:
Hi @morganrallen 
This is fixed in Firmata:
https://github.com/firmata/arduino/pull/438

closed 2019-12-22 by fpistm

morganrallen, 2019-12-22:
Great! thanks!