stm32duino/Arduino_Core_STM32 #843: L011K4

dennyxx82, 2019-12-26:
can add support for L011k4 board

fpistm, 2019-12-28:
Hi @dennyxx82 
I've added this request to the request list #722

closed 2019-12-28 by fpistm

dennyxx82, 2019-12-29:
why my request are closed?

-- 
Cordiali saluti.

ing. Domenico De Natale

DE NATALE ING. DOMENICO
C/DA BONIFACIO
87064 - CORIGLIANO-ROSSANO (CS)
P.IVA: 03311920783
CF.: DNTDNC82T11H490R
CODICE SDI: USAL8PV
tel.: +39 348 1552861
mail.: denataledomenico@gmail.com

fpistm, 2019-12-29:
@dennyxx82 
if you read my previous comment:

> 
> 
> Hi @dennyxx82
> I've added this request to the request list #722

When a new variant is requested I simply add it to the list and close the issue.

dennyxx82, 2019-12-29:
Ok thank’s

Cordiali saluti.

De Natale ing. Domenico
c/da Bonifacio
87064 - Corigliano-Rossano (CS)
P.IVA: 03311920783
CF: DNT DNC 82T11 H490R
CODICE SDI: USAL8PV
cell: +39 348 1552861
fax: +39 0983 82057
mail: denataledomenico@gmail.com