stm32duino/Arduino_Core_STM32 #846: Changing Serial3 Rx pin hangs the sketch

jgromes, 2019-12-30:
**Describe the bug**
When Serial3 Rx pin is changed in the Arduino sketch, the sketch will hang after 64 writes to Serial3 port (likely buffer overflow).

**To Reproduce**
Using the following sketch, serial monitor shows that loop is executed 64 times, then hangs. If `Serial3.setRx(PC5);` line is commented out, it continues to run. Using oscilloscope, it can be seen that without the call to setRx(), Tx line does send the expected data. With call to setRx(), tx line is at 0.

**Workaround**
The only workaround found so far is to change the default USART3 RX pin in PeripheralPins.c, from PB_11 to PC_5.

```c++
HardwareSerial Serial3(USART3);

void setup() {
  Serial.begin(9600);
  Serial3.setRx(PC5);
  Serial3.setTx(PC4);
  Serial3.begin(9600);
}

int i = 0;
void loop() {
  Serial.println(i++);
  Serial3.write('a');
}
```

**Desktop:**
 - OS: [e.g. Windows]
 - Arduino IDE version: 1.8.10
 - STM32 core version: 1.8.0 release
 - Upload method: Mass Storage

**Board:**
 - Name: Nucleo L452RE-P

fpistm, 2019-12-30:
Hi @jgromes 
The **Nucleo L452RE-P** has no RX capabilities on `PC5`:
https://github.com/stm32duino/Arduino_Core_STM32/blob/7704900d51fdc26cfd92c5b35a158e7cb5e3f87d/variants/NUCLEO_L452RE/PeripheralPins.c#L186-L188
While the `Nucleo L452RE` has.

So I guess the HardwareSerial Serila3 is not initialized properly, if you enabled the `CORE_DEBUG`
you would see:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/libraries/SrcWrapper/src/stm32/uart.c#L140

By default using `HardwareSerial Serial3(USART3);` uses the `PB11` and `PB10`.
For RX other possibility is `PC11`.
For TX other possibilities are `PC4` and `PC10`.

Bambofy, 2019-12-30:
no i can confirm this bug still occurs when you compile and upload to the L452RE too.

fpistm, 2019-12-30:
The original issue refers to the `Nucleo L452RE-P`  that's why I set it invalid.
For the `Nucleo L452RE` this will need deeper investigation.

Bambofy, 2019-12-30:
Ah ok, thanks @fpistm. i was testing a few serial options yesterday but i couldn't get any form of HardwareSerial, Serial1/2/3 and SoftwareSerial to communicate to an SIM800L.

The SIM800L's TX pin is connected to STM32's PC4 and the SIM800L's RX pin is connected to STM32's PC5 pin. 

The setup in sketches was as follows:
HardwareSerial Serial3(PC4, PC5)

Also tried the .setRx() .setTx() functions.

Notes are that when using Serial3.flush() it would freezes the entire program.

fpistm, 2019-12-30:
For the  `Nucleo L452RE`:
`HardwareSerial Serial3(PC4, PC5)` is not correct, you reversed the RX and Tx.
It should be:
`HardwareSerial Serial3(PC5, PC4)`

Bambofy, 2019-12-30:
Using this sketch, there is nothing written to Serial3 by the SIM800L.
```

#include "Configuration.h"
#include "PinMap.h"

HardwareSerial Serial3(PC5, PC4);

void setup() {
    Serial.setRx(S_D3);
    Serial.setTx(S_D2);
    Serial.begin(9600);

    Serial3.begin(9600);
    
    Serial.println("Started SIM800L Serial"); 
}
 
/*
 *
 * AT&V = display a long list of configurations.
 */
void loop()
{
	Serial.println("test");

	Serial3.write("AT\r\n");
	delay(2000);
	while (Serial3.available() > 0)
	{
		Serial.println("DATA!!!");
		char c = Serial3.read();
		Serial.print(c);
	}
}
```

jgromes, 2019-12-30:
>The Nucleo L452RE-P has no RX capabilities on PC5

Oops, didn't notice the ifdef condition - sorry about that. With PC11, setRx() works as expected on L452RE-P.

fpistm, 2019-12-30:
I give you any possible Rx/Tx value for the Nucleo-L452RE-P using USART3.
So up to you to use the correct pin mapping.

Bambofy, 2019-12-30:
It doesn't work using the Nucleo-L452RE using USART3 for some odd reason

Bambofy, 2019-12-30:
false alarm sorry, i accidently broke the hardware.

fpistm, 2019-12-30:
OK guys
what is exactly the status?
I'm a little bit confused about issue btw Nucleo L452RE and Nucleo L452RE-P

jgromes, 2019-12-30:
On STM32L452RE-P, it was caused by using incorrect Rx pin.
On STM32L452RE, it was caused by damaged hardware,

Pretty sure we can close this now.

closed 2019-12-30 by jgromes

fpistm, 2019-12-30:
OK thanks for the feedback