stm32duino/Arduino_Core_STM32 #850: Question: A bit of confusion with serials and Variants.

LinoBarreca, 2020-01-03:
Let's do some "brainstorming" together.
I am defining a variant.
I took the template variant.h
There's a section
```
// UART Definitions

// Define here Serial instance number to map on Serial generic name
#define SERIAL_UART_INSTANCE    x //ex: 2 for Serial2 (USART2)
// DEBUG_UART could be redefined to print on another instance than 'Serial'
//#define DEBUG_UART              ((USART_TypeDef *) U(S)ARTX) // ex: USART3
// DEBUG_UART baudrate, default: 9600 if not defined
//#define DEBUG_UART_BAUDRATE     x
// DEBUG_UART Tx pin name, default: the first one found in PinMap_UART_TX for DEBUG_UART
//#define DEBUG_PINNAME_TX        PX_n // PinName used for TX

// Default pin used for 'Serial' instance (ex: ST-Link)
// Mandatory for Firmata
#define PIN_SERIAL_RX           x
#define PIN_SERIAL_TX           x

// Optional PIN_SERIALn_RX and PIN_SERIALn_TX where 'n' is the U(S)ART number
// Used when user instanciate a hardware Serial using its peripheral name.
// Example: HardwareSerial mySerial(USART3);
// will use PIN_SERIAL3_RX and PIN_SERIAL3_TX if defined.
//#define PIN_SERIALn_RX          x // For U(S)ARTn RX
//#define PIN_SERIALn_TX          x // For U(S)ARTn TX
//#define PIN_SERIALLP1_RX        x // For LPUART1 RX
//#define PIN_SERIALLP1_TX        x // For LPUART1 TX
```

I guess that's the correct place to define them.
question 1:
why should I define `PIN_SERIALn_RX` there if I already have the `const PinMap PinMap_UART_TX[]` & `const PinMap PinMap_UART_RX[]` in PeripheralPins.c? Doesn't this create confusion? I guess only one point should be used.

Also.... where should I `#define ENABLE_HWSERIAL5` to say "hey, this variant has a HW serial port so define Serial5"?  I'd expect to do that in the variant.h under the uart definition section...but done there it doesn't work.
At the moment I have to specify it as an extra build flag to make it work.

Thanks and happy new year.

fpistm, 2020-01-03:
Hi @LinoBarreca 
Happy new year too. Thanks

Just a quick note, for those kind of support, please use the forum stm32duino.com.

> question 1:
> why should I define `PIN_SERIALn_RX` there if I already have the `const PinMap PinMap_UART_TX[]` & `const PinMap PinMap_UART_RX[]` in PeripheralPins.c? Doesn't this create confusion? I guess only one point should be used.

An U(S)ART can have several possible pins.
If you instantiate an new Serialx using the peripheral name (ex: `HardwareSerial Serial3(USART3)`) then the first Rx/Tx pin found in the  `const PinMap PinMap_UART_RX/TX[]` in PeripheralPins.c 
will be used.
Defining  `PIN_SERIAL3_RX` and  `PIN_SERIAL3_TX`  allow to force usage of a pin which is not mainly the first one in the array.

> Also.... where should I `#define ENABLE_HWSERIAL5` to say "hey, this variant has a HW serial port so define Serial5"? I'd expect to do that in the variant.h under the uart definition section...but done there it doesn't work.

Well, it works on my side.
I've tested and it works. Defining `ENABLE_HWSERIAL5` allow to `HAVE_HWSERIAL5`:
https://github.com/stm32duino/Arduino_Core_STM32/blob/757493c4ec8f04f9470231a7db5d5ec5873ff7e5/cores/arduino/WSerial.h#L124-L128

then in HardwareSerial.cpp:
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/cores/arduino/HardwareSerial.cpp#L60-L67

Maybe an issue in your env with PIO ?

LinoBarreca, 2020-01-03:
hi @fpistm . Sorry for using the wrong place. It won't happen again :bowing_man: 

> Defining PIN_SERIAL3_RX and PIN_SERIAL3_TX allow to force usage of a pin which is not mainly the first one in the array.

Makes sense.

> Well, it works on my side.
> I've tested and it works.

Yes. I've seen the definition there in WSerial.h.
And I've seen the include...that's why I thought it should have worked...but it doesn't.
Probably something wrong in PIO, as you say. 
Compilation goes fine but I get a linking error about missing Serial3.
Weirdly enough if include it in the general build flags (for every file) I get no errors..
Even if I do a clean build...For sure it's something broken in PIO...I just updated to stm32 6.0.0
I'll revert to 5.7.0 and try again.

Thank you for your time and sorry again :)

fpistm, 2020-01-03:
No worries and welcome.

closed 2020-01-03 by LinoBarreca