stm32duino/Arduino_Core_STM32 #858: [New Variant] STM32G0316-DISCO

DurandA, 2020-01-06:
I'm interested in using STM32duino for the STM32G0316-DISCO which is based on a STM32G0 in SO8 package.

**Board description:**
 - Name: STM32G0316-DISCO
 - Link to all relevant information:
   * [Documentations](https://www.st.com/en/evaluation-tools/nucleo-g071rb.html)
   * [Schematics](https://www.st.com/resource/en/schematic_pack/mb1454-g031j6-b02_schematic.pdf)
   * [STM32G031J6 datasheet](https://www.st.com/resource/en/datasheet/stm32g031j6.pdf)

Whilst I think I could add the variant myself according to [the documentation](https://github.com/stm32duino/wiki/wiki/Add-a-new-variant-(board)), the STM32G031J6 has different pads connected to each pin:
> A multi-bonding approach is used on the small packages in order to offer a maximum of alternate functions and analog inputs. This approach results in multiple die pads connected internally to a single package pin.

I am not sure if specific care should be taken or if it's better to let users use pins as if they were on different pins.

Also, how is pin remapping handled by STM32duino? Maybe @fpistm knows about STM32G0 specifics.

DurandA, 2020-01-06:
I just found out https://github.com/stm32duino/Arduino_Core_STM32/issues/722. I will close this issue if requested but it would be nice if I can get some advices related to pin multi-bonding and pin remapping.

fpistm, 2020-01-07:
Hi @DurandA 
I was not aware about this.
Regarding the documentation:
[image]
It should works but this means that the number of digital pins will be higher then the numbers of physical pins. And it will be to end user to use properly each pins capabilities to avoid any shortcut.

DurandA, 2020-01-07:
Hi @fpistm and thanks for your comment.

What about pin remapping? PA12 can be remapped to PA10 and PA11 can be remapped to PA9 as in STM32G071RB from _NUCLEO_G071RB_. I did not found remapping functions in _NUCLEO_G071RB_ variant. I think that it just won't work if users try to use PA10 (pin 2) or PA9 (pin 8).

fpistm, 2020-01-07:
I think you talk about AF (Alternate Function)
My understanding is that you can use PA12/PA10 to and PA9/PA11 only for one dedicated feature (peripheral). If you want use I2C on physical pin 6 and 5 of the package then you have to choose between  PA12/PA11 or PA10/PA11. They are mutually exclusive:

https://github.com/stm32duino/Arduino_Tools/blob/31f3d43d620878dccbf658f65095f66fec9809b2/src/genpinmap/Arduino/STM32G031J(4-6)Mx/PeripheralPins.c#L46-L48

https://github.com/stm32duino/Arduino_Tools/blob/31f3d43d620878dccbf658f65095f66fec9809b2/src/genpinmap/Arduino/STM32G031J(4-6)Mx/PeripheralPins.c#L57-L58

So PA12/PA11 for I2C2 or PA10/PA11 for I2C1.

DurandA, 2020-01-07:
> I think you talk about AF (Alternate Function)
> My understanding is that you can use PA12/PA10 to and PA9/PA11 only for one dedicated feature (peripheral). If you want use I2C on physical pin 6 and 5 [...]

[image]

I think Alternate Function refers to what function a specific pins perform (i.e. _ADC1_EXTI11_, _ADC1_IN15_, _I2C2_SCL_, ... for PA11) while pin remapping enable a physical pin to map to a different pin:
> Pin pair PA9/PA10 can be remapped in place of pin pair PA11/PA12 (default mapping), using SYSCFG_CFGR1 register.

I examined a generated project from STM32CubeMX and if I remap PA12 to PA10, it adds the line `HAL_SYSCFG_EnableRemap(SYSCFG_REMAP_PA12);` to _stm32g0xx_hal_msp.c_.

fpistm, 2020-01-07:
Right, I've didn't see this.
So for this, it is currently not managed by Arduino API. Anyway, you can call the HAL directly.

DurandA, 2020-01-07:
Thanks. I think the port will be straightforward then.

fpistm, 2020-01-07:
I've open a issue to study that.

DurandA, 2020-01-08:
@fpistm I am in the process of adding this variant and have a few questions:
- Would you recommend to configure the clock to 16 MHz from the high-speed internal RC oscillator (HSI16) or to feed HSI16 to the system PLL to the maximum output frequency of 64 MHz? What is your rule of thumb for other boards; maximum speed?
- There is no serial connected to the built-in ST-Link, only JTCK and JTMS. Is `SERIAL_PORT_MONITOR` mandatory?
- How do serials (e.g. `Serial`, `Serial2` in other boards) in variant.h map to a particular UART interface of the STM32? Is it using the `PinMap_UART_TX[]`/`PinMap_UART_RX[]`? So `Serial` is [0] and `Serial2` is [2] (or [1])?
- I don't see any initialization of the serial interfaces in the other variants. Does the Arduino core expect the defined interfaces such as `SERIAL_PORT_MONITOR` to be readily available?

fpistm, 2020-01-08:
> * Would you recommend to configure the clock to 16 MHz from the high-speed internal RC oscillator (HSI16) or to feed HSI16 to the system PLL to the maximum output frequency of 64 MHz? What is your rule of thumb for other boards; maximum speed?

In a general way, I try to set the maximum speed. But don't worry the clock config is a WEAK function so it can be redefined at sketch level if end user want another clock config.

> * There is no serial connected to the built-in ST-Link, only JTCK and JTMS. Is `SERIAL_PORT_MONITOR` mandatory?

This is only a define used on some example sketch or library. Define it to the generic `Serial` instance would be good.

> * How do serials (e.g. `Serial`, `Serial2` in other boards) in variant.h map to a particular UART interface of the STM32? Is it using the `PinMap_UART_TX[]`/`PinMap_UART_RX[]`? So `Serial` is [0] and `Serial2` is [2] (or [1])?

`Serial` is the generic Serial instance name and it is mapped on the defined `Serialx` instance where **x** is the `U(S)ARTx` peripheral number. Ex: `Serial2` for `USART2`

To define automatically the `Serial` instance, it requires to define in the variant.h the `U(S)ARTx` peripheral number and the default pins Rx/Tx to use::
```
// UART Definitions
#define SERIAL_UART_INSTANCE    2

// Default pin used for 'Serial' instance
#define PIN_SERIAL_RX           PA3
#define PIN_SERIAL_TX           PA2
```

Ten it will be instantiate and when you do the Serial.begin then it check in the `PinMap_UART_TX[]`/`PinMap_UART_RX[]` the propre pins config for Rx and Tx.

> * I don't see any initialization of the serial interfaces in the other variants. Does the Arduino core expect the defined interfaces such as `SERIAL_PORT_MONITOR` to be readily available?

no, the core does not use `SERIAL_PORT_MONITOR` but for Arduino compatibility it is fine to define it as told in point 1.

commit e7bb20b0 (2020-02-04): Add STM32G0316-DISCO

Fixes #858
closed 2020-02-04 by fpistm