stm32duino/Arduino_Core_STM32 #859: LoRa->RAK811 pins configuration

RAKWireless, 2020-01-07:
Power on for LoRa chip:
`  **pinMode(RADIO_XTAL_EN,OUTPUT);  //Power LoRa module
  digitalWrite(RADIO_XTAL_EN,1);**`

External antenna receiving and transmitting state switching pin configuration:
Receive state:
`
  **pinMode(RADIO_RF_CRX_RX,OUTPUT);  
  digitalWrite(RADIO_RF_CRX_RX,1);  //control LoRa work to receive
  pinMode(RADIO_RF_CTX_PA,OUTPUT);  
  digitalWrite(RADIO_RF_CTX_PA,0);  //**
`
Transmit state:
`
  **pinMode(RADIO_RF_CRX_RX,OUTPUT);  
  digitalWrite(RADIO_RF_CRX_RX,0);  
  pinMode(RADIO_RF_CTX_PA,OUTPUT);  
  digitalWrite(RADIO_RF_CTX_PA,1);  //control LoRa send by PA_BOOST**
`

fpistm, 2020-01-07:
Hi @RAKWireless 
What is the issue ? What did you expect?
Is it for an example ? If yes I guess it could be fine to add them in the Wiki?

RAKWireless, 2020-01-07:
Hi fpistm,

Sorry for confuse.
We've ever used this BSP on RAK811 LoRa board several days ago, but we found that the RSSI is very weak, for example, the node is only 10m away from the gateway, the RSSI is almost -40...
After debuging, we've fixed this issue. Actually, it is not because BSP, users just should define some pins in their application.
The content we just paste is just used to remind users to define in their application code, when they use RAK811 LoRa board with this Arduino BSP.
More details can be found here:
https://github.com/RAKWireless/RAK811_LoRaWAN_Arduino

fpistm, 2020-01-07:
@RAKWireless 
What I can propose is to add a note in the Wiki and also reference your repo in it and in the README.md (in the comment field of the RAK811 line).

RAKWireless, 2020-01-07:
OK, i think that will be helpful for users.
Thank you!

fpistm, 2020-01-07:
OK done.
https://github.com/stm32duino/wiki/wiki/Connectivities#lora

Do not hesitate to update it if needed.

commit ea866457 (2020-01-07): Update README.md

Fixes #859
closed 2020-01-07 by fpistm