stm32duino/Arduino_Core_STM32 #862: USB Serial not working with HAL UART

Tretgo, 2020-01-08:
Preconditions:
OS: Win10
Arduino 1.8.10
STM32 1.8.0
BluePill F103
HID Bootloader
USB CDC Serial

USB Serial output is not working when setting
HAL_UART_MODULE_ONLY
HAL_UART_MODULE_ENABLED
in build_opt.h

Target: LIN Serial implementeation on UART1 and USB Serial for debugging

Root cause:
https://github.com/stm32duino/Arduino_Core_STM32/blob/ea8664573c69acf7673103b6ef198ef4b22d00b9/cores/arduino/Print.cpp#L199

Function write in Print.cpp gets disabled when setting UART into HAL mode.
But only uart_debug should be disabled. 

Working bugfix:

`extern "C"
{
  __attribute__((weak)) int _write(int file, char *ptr, int len)
  {
#if defined(HAL_UART_MODULE_ENABLED) && !defined(HAL_UART_MODULE_ONLY)
    switch (file)
    {
    case STDOUT_FILENO:
    case STDERR_FILENO:
      uart_debug_write((uint8_t *)ptr, (uint32_t)len);
      break;
    case STDIN_FILENO:
      break;
    default:
      ((class Print *)file)->write((uint8_t *)ptr, len);
      break;
    }
#else
    if (file != STDERR_FILENO)
      ((class Print *)file)->write((uint8_t *)ptr, len);
#endif
    return len;
  }
}`

fpistm, 2020-01-09:
Hi @Tretgo 
I guess this is when you use `printf()` with SerialUSB ?

Tretgo, 2020-01-12:
Hi,
just checked.
`println()` works, but `printf()` does not.

PR #867 (2020-01-15): Print::printf() generic
Before it was only linked to U(S)ART while `Print` is a Base class.

Fixes #862
commit c4b96a78 (2020-01-15): Print::printf() generic

Before it was only linked to U(S)ART while Print is a
Base class.

Fixes #862
closed 2020-01-15 by fpistm