stm32duino/Arduino_Core_STM32 #865: Error in Uploading STM32F401RE

Vicky-S, 2020-01-11:
Desktop (please complete the following information):
 - OS: [e.g. Windows]
 - Arduino IDE version: [e.g. 1.8.8]
 - STM32 core version: [e.g. 1.8.0]

Board (please complete the following information):
 Name: Nucleo 401RE

I updated my STM32 official core today and after this i am getting the following error if I try to upload the code in Arduino. Please check below for the error shown in arduino IDE.

Insufficient disk space on current disk.
Insert another disk and type <Return> to continue... An error occurred while uploading the sketch

fpistm, 2020-01-11:
Seems strange.
Did you try to update the STLink Firmware?

I guess this is with MassStorage ?
Which sketch you use ?

Vicky-S, 2020-01-11:
I can create HEX file and upload it via STLINK UTILITY. But I can't directly program in arduino.

Even blink example giving the same error while uploading.

fpistm, 2020-01-11:
When I've got this error it was due to an issue with STLink Firmware. So please try to update it.

Vicky-S, 2020-01-11:
Ok...

fpistm, 2020-01-11:
If this not solved your issue share all option you used, the STLink Fw version and all relevant information.
Anyway as the Hex file works when flashed with STLink utility means the FW is well formed and rely on massStorage copy. And MassStorage copy is managed by STLink FW.

fpistm, 2020-01-13:
Hi @Vicky-S 
I've made some tests (Linux/Windows, MassStorage, STlink SWD)  on my Nucleo F401 and I have no issue.

Vicky-S, 2020-01-13:
I removed and Reinstalled the STM32 core in Arduino Boards manager. Problem is solved.

closed 2020-01-13 by Vicky-S