stm32duino/Arduino_Core_STM32 #868: STM32 recompiles every time i build

Bambofy, 2020-01-13:
When reporting any issue, please try to provide all relevant information:

**Describe the bug**
When i build/verify a sketch that uses the STM32 board nucleo-64 L452RE, the STM32 is always rebuilt causing the time taken to verify a sketch to be very long.

**To Reproduce**
1. Setup STM32 l452re board.
2. Compile a blank sketch.
3. See stm32 console building e.g

> ""C:\Users\Richa\AppData\Local\arduino15\packages\STM32\tools\xpack-arm-none-eabi-gcc\9.2.1-1.1\bin\arm-none-eabi-g++" -mcpu=cortex-m4 -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mthumb "@C:\Users\Richa\AppData\Local\Temp\VMBuilds\SAINSoftware\STM32_Nucleo_64\Debug/build_opt.h" -c -Os -w -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DVM_DEBUG_BREAKPAUSE -DVM_DEBUGGER_TYPE=VM_DEBUGGER_TYPE_GENERIC_OBJECT -DVM_DEBUGGER_SOFT_TRANSPORT_WRITER=Serial -DVM_DEBUGGER_SOFT_TRANSPORT=Serial -DVM_DEBUG_BANDWIDTH_THROTTLE_MS=33 -DVM_DEBUG -DVM_DEBUG_ENABLE=1 -DVM_DEBUGGER_TYPE_GENERIC_OBJECT=21 -DVM_DEBUGGER_TYPE_NO_SERIAL=20 -DVM_DEBUGGER_TYPE_MS430_SERIAL_=19 -DVM_DEBUGGER_TYPE_SERIALUSB=18 -DVM_DEBUGGER_TYPE_USBAPI=17 -DVM_DEBUGGER_TYPE_NET_UDP=16 -DVM_DEBUGGER_TYPE_HARDWARESERIAL3=15 -DVM_DEBUGGER_TYPE_HARDWARESERIAL2=14 -DVM_DEBUGGER_TYPE_HARDWARESERIAL1=13 -DVM_DEBUGGER_TYPE_CDCSerialClass=12 -DVM_DEBUGGER_TYPE_COSA=11 -DVM_DEBUGGER_TYPE_Uart=10 -DVM_DEBUGGER_TYPE_NET_CONSOLE=9 -DVM_DEBUGGER_TYPE_TTYUART=8 -DVM_DEBUGGER_TYPE_USBSERIAL=7 -DVM_DEBUGGER_TYPE_USART=6 -DVM_DEBUGGER_TYPE_UART=5 -DVM_DEBUGGER_TYPE_TEENSY=4 -DVM_DEBUGGER_TYPE_USB=3 -DVM_DEBUGGER_TYPE_FASTSERIAL=2 -DVM_DEBUGGER_TYPE_SOFTWARESERIAL=1 -DVM_DEBUGGER_TYPE_HARDWARESERIAL=0   "-IC:\Users\Richa\Documents\GitHub\Remote-Train-Sensor\SAIN Software - Visual Micro\SAINSoftware\SAINSoftware" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\cores\arduino/avr" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\cores\arduino/stm32" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\cores\arduino/stm32/LL" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\cores\arduino/stm32/usb" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\cores\arduino/stm32/usb/hid" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\cores\arduino/stm32/usb/cdc" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\system/Drivers/STM32L4xx_HAL_Driver/Inc" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\system/Drivers/STM32L4xx_HAL_Driver/Src" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\system/STM32L4xx" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -w -x c++ -E -CC -DSTM32L4xx -DARDUINO=108010 -DARDUINO_NUCLEO_L452RE -DARDUINO_ARCH_STM32 -DBOARD_NAME="NUCLEO_L452RE" -DSTM32L452xx -DHAL_UART_MODULE_ENABLED "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\tools\CMSIS\5.5.1/CMSIS/Core/Include/" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\system/Drivers/CMSIS/Device/ST/STM32L4xx/Include/" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\system/Drivers/CMSIS/Device/ST/STM32L4xx/Source/Templates/gcc/" "-IC:\Users\Richa\AppData\Local\arduino15\packages\STM32\tools\CMSIS\5.5.1/CMSIS/DSP/Include" -I"C:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\cores\arduino" -I"C:\Users\Richa\AppData\Local\arduino15\packages\STM32\hardware\stm32\1.8.0\variants\NUCLEO_L452RE" "C:\Users\Richa\AppData\Local\Temp\VMBuilds\SAINSoftware\STM32_Nucleo_64\Debug\SAINSoftware.cpp" -o "nul""

4. Recompile the sketch and the calls above run again.

**Expected behavior**
Once the STM32 package is compiled it shouldn't need to re-compile every build since it doesn't change.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - Windows
 - 1.8.10
 - STM32 core version 1.8.

**Board (please complete the following information):**
 - Nucleo-64 L452RE

**Additional context**
This issue is the same as here: https://github.com/stm32duino/Arduino_Core_STM32/issues/770
However there is no setting "Aggressively cache compiled core"  in the arduino 1.8.10 IDE.

See attached file for full logs
[stm32l452re successful build.txt](https://github.com/stm32duino/Arduino_Core_STM32/files/4054575/stm32l452re.successful.build.txt)
[STM32 building every time.txt](https://github.com/stm32duino/Arduino_Core_STM32/files/4054576/STM32.building.every.time.txt)

mrguen, 2020-01-13:
Caching might not be active if it was de-activated before 1.8.10 and you upgraded. The option is no more displayed in the IDE, but you can change the preferences.txt.

I also notice that even with this option active it tends to recompile often. At least everytime you changed one option and if you go back to an old set of option you used the same day, it will recompile anyway. Plus it is extremely slow, but probably this is because of a gcc bug to be fixed in next version.

Bambofy, 2020-01-13:
> Caching might not be active if it was de-activated before 1.8.10 and you upgraded. The option is no more displayed in the IDE, but you can change the preferences.txt.
>
> I also notice that even with [...]

I have set the option "compiler.cache_core=true"  inside the perferences.txt however this didn't seem to fix anything.

mrguen, 2020-01-14:
I am waiting for the next Arduino IDE realease since it's a paint to work on my project now. It sometimes takes 5 or 10 or more minutes to compile. "Hooray Python" maybe?

fpistm, 2020-01-14:
@Bambofy, @mrguen 
As I said several times, I could do anything on this.
This is due to how Arduino handles the build.
So stop to claims here it is slow.
@Bambofy about your log what I can see is your sketch is dependant to:

1st log:
`vm.platform.root.path=c:\program files (x86)\microsoft visual studio\2019\preview\common7\ide\extensions\p1hbfuhp.cmm\Micro Platforms\arduino16x`
2nd log:
`vm.platform.root.path=c:\program files (x86)\microsoft visual studio\2019\preview\common7\ide\extensions\safxk1c5.4tc\Micro Platforms\arduino16x`

I don't know exactly your environment (I gues VScode)  nor the sketch you built.
Anyway, like this path change, I guess all is rebuilt.
In the Arduino IDE, if you change one option, the IDE rebuild all.
Under Linux the time to build is 10x faster (and more) for the same core and toolchain version.
Under Windows, I know several network access are made for example slowing down the build and this is worst when it is a third party core.

So, it is not a core issue, I close this issue.

closed 2020-01-14 by fpistm