stm32duino/Arduino_Core_STM32 #869: [New Variant] Can Filter with STM32F105RBT6

EliasKotlyar, 2020-01-13:
**Board description (please complete the following information):**
 - Name: Can Filter
 - Link to all relevant information:
   * Documentations
None available because its a "noname" product. I have completed a short overview here:
https://github.com/EliasKotlyar/Canfilter

   * Schematics
None available, however the board itself is pretty basic.

   * Where this hardware can be purchased
Search on Aliexpress for "Can Filter"

* Why are you using this device?
Dont event ask...I want to retrofit a CAN-Device from a newer Car(Idrive from a F10) into my old car. Both Devices are using different commands and CAN-Speeds. First i wanted to make a board myself, however i stumbled across this board and bought it for 10 Dollar.

EliasKotlyar, 2020-01-13:
I must say that the integration itself is working, and the device can be programmed trough Arduino. However there is one critical issue which i cannot figure out:
-If i setup the blink example, it does not work
-If i setup a simple LED on/off example, its working like expected
-A blink example in SW4STM is working fine without any problems

I suppose that its a timing issue and thereis something missing in the configuration, but i cannot figure out what exactly. The SystemClock_Config function is same on both SW4STM and Arduino. Also the HSE_VALUE is set to 25000000U in Arduino.

fpistm, 2020-01-14:
Hi @EliasKotlyar 
For the HSE issue, simply define the correct HSE value in the variant.h as by default it is 8000000.

For your variant, it should be better to create a generic one for STM32F105RBT6  and then declare in boards.txt the dedicated one CAN Filter.

EliasKotlyar, 2020-01-14:
Hello @fpistm 

I have defined the correct one(HSE: 25000000U) in variant.h here:
https://github.com/stm32duino/Arduino_Core_STM32/pull/870/files#diff-66041f938b1048c240784af4ac5d57c6R22

However it does not solve the problem. Blink is not working, normal LEDs on/off are working fine.

I suppose it may be also some bug/quirk of the "Connectivity Line" SOCs, which STM32F105 belongs to. There are no other Boards defined supporting this Product-line, so its maybe something SOC specific which needs a certain patch. The interesting part is, that the SW4STM is working fine with this configuration

fpistm, 2020-01-14:
OK @EliasKotlyar 
I will review it deeply and then go back to you.

fpistm, 2020-02-14:
Request added to the request list: #722

closed 2020-02-14 by fpistm

XGhostX33, 2023-11-03:
Hello how to connect this board to PC? to use Can Hacker