stm32duino/Arduino_Core_STM32 #871: I2S Interface? HAL?

Bambofy, 2020-01-14:
Hi, is there a way to use the I2S library with STM32Duino?

Thanks
- Richard

fpistm, 2020-01-14:
Hi @Bambofy 
currently there is no "Arduino" library for I2S.
You can use HAL or LL at sketch level by enabling the `HAL_I2S_MODULE_ENABLED`.

See https://github.com/stm32duino/wiki/wiki/HAL-configuration

Bambofy, 2020-01-14:
If i #define HAL_I2S_MODULE_ENABLED in stm32l4xx_hal_conf_default.h would that do anything? i cannot see any ifdefs that include specific header files?

fpistm, 2020-01-14:
Use the hal_conf.h to define it at sketch  level. Then you will have access to all HAL I2S API.

fpistm, 2020-01-14:
@Bambofy 
For this kind of question, please use the stm32duino.com forum.

Bambofy, 2020-01-16:
Hi, I have been looking through the stm32l4 drivers and i can't find the i2s driver? It is included in other versions but not this one?

fpistm, 2020-01-16:
Simply because there is no I2S on L4...
https://www.st.com/en/microcontrollers-microprocessors/stm32l4-series.html

fpistm, 2020-01-17:
@Bambofy 
when I replied I did not see you talk about L4.
FYI, all HAL/LL are provided so if you do not find a HAL module this simply means the feature is not available for this serie.
I close this one as I consider it as answered.

closed 2020-01-17 by fpistm