stm32duino/Arduino_Core_STM32 #873: Improvement: Enable -g on all builds

matthijskooijman, 2020-01-17:
Currently, the `-g` compiler option for adding debug symbols is only passed when the optimization level is "Debug". However, I think that passing -g is *always* a good idea, even with higher optimization levels.

Of course, when you expect to be debugging, you can simply enable the debug optimization level for a build. However, sometimes you need to debug a production build, in particular when you cannot reproduce the problem using a debug build, or when you want to analyze a stack dump collected from some production system.

It is true that aggressive optimizations might interfere with debugging, but in practice debugging still works reasonably well (and certainly better than having *no* debugging symbols at all).

Also, AFAIK adding debug symbols does not affect the produced machine code, so the only downside of adding debug symbols might be that compilation is a little slower.

Implementing this would, I think, be a matter of removing the `-g` option from the Debug optimization levels in `boards.txt` (and replacing `-g` with `-Og` in its description, which is a good idea regardless). Then, the `-g` option can be added unconditionally in `platformt.txt`.

fpistm, 2020-01-17:
Hi @matthijskooijman 
I've already had several discussion about this and adding `-g` by default was not desired.
A better approach could be to remove the `-g` from the 'Debug' optimization and add a menu to enable it or not.
The main drawback is the menu addition (huge boards.txt and new menu)

matthijskooijman, 2020-01-17:
> I've already had several discussion about this and adding -g by default was not desired.

Is there any record of this discussion you could link to? I'm curious what the arguments against are.

> A better approach could be to remove the -g from the 'Debug' optimization and add a menu to enable it or not.
The main drawback is the menu addition (huge boards.txt and new menu)

Yeah, that could be an alternative. I guess Arduino should really support platform-level menus rather than just board-level menus, but that is probably difficult to implement wrt backward compatibility...

fpistm, 2020-01-17:
Well don't remember where it was discussed (GitHub, forums,...)
I'm not against add '-g', I've only related what I remembered.

lyusupov, 2020-01-17:
AFAIR, -g increases binary size even if -Os is activated. Can be very critical for MCUs with small flash memory size.
It also increases build host's RAM comsumption and overall build time.
IMHO, setting of -g by default, in general, does not seem to be a good idea.

matthijskooijman, 2020-01-17:
> AFAIR, -g increases binary size even if -Os is activated. Can be very critical for MCUs with small flash memory size.

It will certainly increase the .elf file size, but I would expect it does not affect the actual code size (though I could not find anything definitive about this).

>  It also increases build host's RAM consumption and overall build time.

Yeah, that is a downside, but I would think it is a minimal cost.

VitorBoss, 2020-02-18:
I have done some tests with [Speeduino code](https://github.com/noisymime/speeduino/) on a Black 407VE, here ar the results:

`          |   -g    |   -Os   |   -O1   |   -O2   |   -O3   |
BIN size  | 143.716 | 114.972 | 140.436 | 130.768 | 142.436 |
ELF size  | 404.322 | 276.396 | 341.632 | 272.688 | 335.516 |
RAM use   |  10560  |  10560  |  10560  |  10560  |  10560  |
free RAM  |  50691  |  50711  |  50675  |  50703  |  50711  |
loops/S   |  7196   |  6995   |  7446   |  7702   |  7716   |`

matthijskooijman, 2020-02-18:
@VitorBoss, when you say -g, that means the current `-g -Og` flags I think? Are there any conclusions that you want to connect to your results? I guess it might be interesting to see the results of various `-O` options with *and* without `-g` added, which might provide some info on whether or not to enable `-g` by default?

matthijskooijman, 2020-02-18:
Also, thanks for taking the time to provide some test data! :-)

VitorBoss, 2020-02-18:
I'm using only the menu options

VitorBoss, 2020-02-18:
Using **-g -Os** give almost same results from **-Os** except for ELF size, it increased to 1.648.196 bytes

fpistm, 2020-10-16:
I think I will add a new menu to enable or not the debug.
Like this it can be enable for all Ox level and by default it will be disabled for backward compatibility.

any feedback ar welcome on this proposal :wink:

fpistm, 2021-07-13:
As no feedback, I will add the new menu.

PR #1446 (2021-07-14): config: add debug information menu
This PR add a new menu `Debug information`.
By default it is not enabled. Enable it will produce debugging information.

[image]

It also fix the `Optimize` menu which displayed `Debug (-g)` instead of "Debug (-Og)".

Fixes #873

For further reading: https://gcc.gnu.org/onlinedocs/gcc/Debugging-Options.html#Debugging-Options

/CC @matthijskooijman
commit 96955e2f (2021-07-14): config: add debug information menu

Fixes #873
closed 2021-07-14 by fpistm