stm32duino/Arduino_Core_STM32 #875: Firmata: Analog inputs are not enumerated anymore on NUCLEO476

jmchiappa, 2020-01-19:
**Describe the bug**
Analog input are no more enumerated when using NUCLEO 476RG. I'm using the firmata_test application. Before the 1.8.0, everything works well.

**To Reproduce**
Compile StandardFirmataPlus.ino on NUCLEO 476RG and flash it.
Open firmata_test.
Observe that, at the end, no more analog input appears.

**Expected behavior**
analog inputs must be enumerated and activated as soon as the application starts.

**Desktop (please complete the following information):**
 - OS: linux mint 19.3
 - Arduino IDE version: 1.8.9
 - STM32 core version: 1.8.0
- Firmata : 2.5.8

**Board (please complete the following information):**
 - Name: NUCLEO L476RG
 - Hardware Revision: C-04

**Additional context**
it seems to be related to the new analog pins alignment.
Once Firmata server (stm32 side) receives a CAPABILITY REQUEST, it parses all pins (**all** means TOTAL_PINS which is defined as NUM_DIGITAL_PINS.
On NUCLEO L476RG, NUM_ANALOG_FIRST not defined. So it wil be assigned to `#define NUM_ANALOG_FIRST          (NUM_DIGITAL_PINS + 1)`, meaning that A0 is assigned to TOTAL_PINS+1, A1 = TOTAL_PINS+2, and so on...
as `#define IS_PIN_ANALOG(p)        ((p >= A0) && (p < (A0 + TOTAL_ANALOG_PINS)) && !pinIsSerial(p))` in Firmata/Boards.h, p is always below A0
So, analog pins are never scanned in the firmata application.

fpistm, 2020-01-19:
Hi @jmchiappa 
I've made a PR in the firmata repo which has been merged but not released yet.
Try the master of the Firmata library.

jmchiappa, 2020-01-19:
Hi @fpistm ,
 ok, I've a try on firmata master. From a clean git clone, it doesn't work too. I think there's a little fix to do in your firmata PR :
`#if !defined(STM32_CORE_VERSION) || (STM32_CORE_VERSION  <= 0x01080000)`
to
`#if !defined(STM32_CORE_VERSION) || (STM32_CORE_VERSION  < 0x01080000)`

so that it will use the new analog definition.
With this fix, it works again, thanks

fpistm, 2020-01-19:
Ah you're right.
I have to do a new PR on firmata.

closed 2020-01-19 by jmchiappa

fpistm, 2020-01-20:
For ref:
https://github.com/firmata/arduino/pull/443