stm32duino/Arduino_Core_STM32 #876: STM32F411CE doesn't seem to work

TzOk83, 2020-01-19:
I've tried to upload BLINK example (I've changed LED_BUILTIN to PC13) using "Generic F411CE" Board profile to the "Black Pill" WeAct V2.0 board (STM32F411CEU6), and it doesn't seem to work. I've also tried turning the USB support as CDC Generic Serial, but there was no serial port after uploading this firmware. I've tried uploading either by SWD (ST-Link v2) or by USB-DFU. Both times the upload was successful, but program didn't seem to start (of course I have tried to reset and power cycle the MPU).

fpistm, 2020-01-19:
Hi @TzOk83 
it seems a clock config issue.
You can try to reconfigure it defining the `SystemClock_Config `at sketch level:
```
extern "C" void SystemClock_Config(void)
{
...
}
```

flatpedals, 2020-01-19:
I have a WeAct 1.3 411CE and I can upload in Arduino IDE and PlatformIO via DFU (STM32Cube prog) and Serial(USB to TTL dongle). 
ArduinoIDE: 'Blackpill F401CC' under 'Generic STM32F4 series'
[image]

PlatformIO: edited 'BlackPill 401CC (ST)' to a 'BlackPill 411CE (ST)'. 
[blackpill_f411ceJSON(PIO).zip](https://github.com/stm32duino/Arduino_Core_STM32/files/4083534/blackpill_f411ceJSON.PIO.zip)

As I said DFU and Serial are working, then there is a working Serial Port available (STM Serial) while uploaded sketch is running. 

I updated the core and the new 'Generic F411CC' showed up in board manager for a short while but now it's gone, so I can't select it to see if it's a problem vs the 'Blackpill F401CC'.

All good with this board untill now, maybe a STM32DUINO Bootloader would be nice so BOOT0 isn't required at every upload. I also soldered a Winbond 25q32jv and the SMD cap but never got to test that SPI memory.

fpistm, 2020-01-20:
@flatpedals 
the Generic F411Cx has been added recently and is merged in the master: #866

MCUdude, 2020-01-20:
By looking at the pictures of the F411C "Black Pill" board on [this Aliexpress listing](https://www.aliexpress.com/i/4000138305460.html) It appears to me that the external crystal is running at 25 MHz. The generic pinouts expect an 8 MHz crystal/clock.

You can try to change the value of [this line](https://github.com/stm32duino/Arduino_Core_STM32/blob/a37f9590e6752076068f893280ee88fe3d456f36/variants/Generic_F411Cx/variant.cpp#L125) to 25 (actual input clock frequency in MHz) and see if it makes a difference. Oh, you'll have to do a manual install since 1.9.0 isn't out yet.

fpistm, 2020-01-21:
@MCUdude 
In fact there is issue with the generic systemCoreClock config.
It configures the clock assuming it is a 8Mhz for HSE but by default it is defined to 25MHz
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/STM32F4xx/system_stm32f4xx.c#L68

So it is misaligned when HSE is not bypass.

MCUdude, 2020-01-21:
> In fact there is issue with the generic systemCoreClock config.
It configures the clock assuming it is a 8Mhz for HSE but by default it is defined to 25MHz
https://github.com/stm32duino/Arduino_Core_STM32/blob/master/system/STM32F4xx/system_stm32f4xx.c#L68

> So it is misaligned when HSE is not bypass.

But it would have worked if he used an extern 8 MHz crystal instead, and not touching the code?

MCUdude, 2020-01-21:
The board uses a crystal oscillator (that needs to be driven), not a clock (that drives itself). I will assume that since the MCU detects a crystal it will not enable bypass mode, and thus nothing will work since the expected clock frequency is 8 MHz and not 25.

What I do want to know is that if there's something that can be done to the pinout's clock config that somehow can detect what frequency is being applied, and adjust its setting based on this. Can this be done? Can the clock config functions on the generic pinouts be further improved to automatically support a wider range of external clocks/crystals without user interaction?

TzOk83, 2020-01-21:
Haven't tried uploading yet, but I believe this should work:
[PILL_F411XX.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/4090272/PILL_F411XX.zip)

```
# BlackPill F411CE
GenF4.menu.pnum.BLACKPILL_F411CE=BlackPill F411CE
GenF4.menu.pnum.BLACKPILL_F411CE.upload.maximum_size=524288
GenF4.menu.pnum.BLACKPILL_F411CE.upload.maximum_data_size=131072
GenF4.menu.pnum.BLACKPILL_F411CE.build.board=BLACKPILL_F411CE
GenF4.menu.pnum.BLACKPILL_F411CE.build.product_line=STM32F411xE
GenF4.menu.pnum.BLACKPILL_F411CE.build.variant=PILL_F411XX
```

fpistm, 2020-01-21:
In fact you want a "Clock framework". 😉 
Anyway this is something, I will not be able to do.
My guess, is that we cold not handle easily all possible value and they will be always complains about that. The generic should only implement a default clock config based on the HSI/MSI/CSI/LSI.
As a generic variant the end user would have to use his own `SystemClock_Config` if he would customize it. The generic are mainly for those who want test quickly a custom board.
A "commercial" board could then be added easily using the generic one.

As I explain in an other issue, several variants could now point to the generic one with some customization under board switch.

TzOk83, 2020-01-21:
With the attached BlackPill 411CE profile it uploads ok, and BLINK example works, but I can't make USB serial to work.

fpistm, 2020-01-21:
@TzOk83 
You have to ensure the USB clock is at 48 MHz

TzOk83, 2020-01-21:
Have no idea how to check this. The crystal is 25 MHz, and MCU in this version is supposed to run at 100 MHz, yet I took most of the files from PILL_401XX which runs at 84 MHz.

fpistm, 2020-01-21:
The simpliest way is to use cube MX to generate a proper config.

TzOk83, 2020-01-21:
If someone is knows how to use, then probably it is... so if you have Cube MX and can use it, could you be so nice and do this?

flatpedals, 2020-01-22:
@TzOk83
For USB serial to work check the USB CDC option as I mentioned. Also check the board schematic on WeAct github to see which pins they used for USB and make sure you are not using them for other stuff. 
I am successfully using the 411CE v1.3 with the edited platformio.ini and board def.

TzOk83, 2020-01-22:
I have checked it. When it is checked I have a problem with entering BOOT0 bootloader (CDC), usually I get USB device fault (but in normal mode virtual serial port is visible in the system). Previously I was using old "BluePill", so more or less I know how to set the options in Arduino. Could you upload here your modified 411CE profile / boards.ini?

https://github.com/jvitkauskas/weact/tree/master/STM32F411CEU6%20black%20board
Did you mean this GitHub page?

flatpedals, 2020-01-22:
In Arduino IDE I didn't edit anything, I am just using the 'Black pill 401CE' profile. See may first comment here, I posted the edited JSON  and .ini only for PlatformIO. 
You could also try stevestrong's core:  https://github.com/stevstrong/Arduino_STM32, he has done some specific work for the F4. 
My 411CE works with either core.
I suggest uninstalling every related driver if you are on Win10 (show hidden devices in device manager and go clean).

fpistm, 2020-01-22:
With this core, no specific driver are needed as we use the STMicroelectronics PID.

TzOk83, 2020-01-22:
I have roughly checked it and it seems to work now.

%LOCALAPPDATA%\Arduino15\packages\STM32\hardware\stm32\1.8.0:
[PILL_F411XX.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/4100339/PILL_F411XX.zip)

boards.txt:
```
# BlackPill F411CE
GenF4.menu.pnum.BLACKPILL_F411CE=BlackPill F411CE
GenF4.menu.pnum.BLACKPILL_F411CE.upload.maximum_size=524288
GenF4.menu.pnum.BLACKPILL_F411CE.upload.maximum_data_size=131072
GenF4.menu.pnum.BLACKPILL_F411CE.build.board=BLACKPILL_F411CE
GenF4.menu.pnum.BLACKPILL_F411CE.build.product_line=STM32F411xE
GenF4.menu.pnum.BLACKPILL_F411CE.build.variant=PILL_F411XX
```

fpistm, 2020-01-23:
> @MCUdude
> In fact there is issue with the generic systemCoreClock config.
> It configures the clock assuming it is a 8Mhz for HSE but by default it is defined to 25MHz [...]

I'm wrong, the default is 25MHz as @MCUdude note in the 
https://github.com/stm32duino/Arduino_Core_STM32/blob/4e1c22561ae15c976a42699c97e08c53dbb3f4b7/system/STM32F4xx/system_stm32f4xx.c#L67-L69

fpistm, 2020-02-03:
The clock config issue from the Generic F411Cx has been solved in #905.
So I close this one.
The BlackPil F4x1 support will be added reviewed to use the Generic F4x1 variant.

closed 2020-02-03 by fpistm

faelpinho, 2020-04-19:
Hello. Do you know when will have available a new version from this lib? I need to use Generic F411CE. I've tried all tips here, but the upload to this board is not working yet.
Thanks!

fpistm, 2020-04-19:
I'm currently doing a full validation.
It should be release in the coming weeks.

Onefabis, 2020-07-08:
Serial steel doesn't work. If I try to open Arduino Monitor and it show message that 'Board COM3 is unavailable'. However, blink example works fine and my custom projects works well. But Encoder with custom library work ugly, something wrong, perhaps with Clock frequency, but I can't even debug this with Serial monitor, because it can't open properly.

fpistm, 2020-07-08:
Hi @Onefabis 
without more details it is hard to help you.
which core? What did you select in the menu? what is your board and clock config? ...

Onefabis, 2020-07-08:
Hi, @fpistm, what did you mean about the core? Just got WeAct STM32F411CEU6 board. Didn't change anything, so I assume that it use 25MHz as a default clock speed. Here are settings in Arduino IDE (Stm32 Boards is v1.9.0)
[image]

fpistm, 2020-07-08:
Well, for this boards clock is 25MHz.
with your settings, the Serial instance mapped to Serial1 which is an HardwareSerial on USART1 (PA10/PA9) so you have to connect a FTDI or a real UART connector on those pins.
If you want use the Serial overs USB then you have to select USB support: CDC (generic 'Serial' supersede U(S)ART).

For this kind of support please use the forum: https://www.stm32duino.com

samveen, 2020-07-11:
@flatpedals 

> The STM32 HID seems to allow successful flashing but the MCU is never reset and the sketch doesn't run. I have to erase memory to get it out of that frozen state.
Can you tell me the method you used to erase all memory? I seem to be stuck in a similar situation: my Blackpill F411 only works in DFU mode, no matter what I try to program using `dfu-util`

Olareanu, 2021-05-13:
> build_flags =
> -D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
> -D USBCON
> -D USBD_VID=0x0483
> -D USB_MANUFACTURER="Unknown"
> -D USB_PRODUCT=""BLACKPILL_411CE""
> -D HAL_PCD_MODULE_ENABLED`

This worked for me (Serial trough USB port and blinking LED). It looks like only PIO_FRAMEWORK_ARDUINO_ENABLE_CDC and USBCON are needed. Using weAct 3.0 Black pill STM32F411CE with this PIO config:

[env:genericSTM32F411CE]
platform = ststm32
board = blackpill_f411ce
framework = arduino
monitor_speed = 115200
upload_protocol = dfu
build_flags = 
	-D PIO_FRAMEWORK_ARDUINO_ENABLE_CDC
	-D USBCON

pitass82, 2021-10-15:
[image]
Serial worked initiallz for me too, I made it work with:
1) editing code: serial begin in brackets 9600 means Serial.begin(9600); instead Serial.begin(); (open serial monitor w same baudrate
2) plus setting arduino IDE Tools : CDC no generic serial
[image]
Hope it helps

miguelwon, 2022-02-16:
Hi,

I just received a brand new BlackPill STM32F411CUE6 and I'm having a related problem. I trying to upload the blink example with Arduino IDE. I'm uploading through USB and using DFU mode. I'm able to upload only in case I set to `None` the `USB support`. The moment I set it to any of the CDC options, I get the `Target not found `error.

I also have a BlackPill STM32F401C and it works fine.

fpistm, 2022-02-16:
Hi @miguelwon,

Upload method  and USB support menu has no link.
While your board is properly reset in DFU bootloader mode upload should work as expected.
I've tested several time and I have no issue.
1. press Boot0 button then plus USB cable or press NRST
2. press upload

The error `Target not found` is raised by the STM32CubeProgrammer if no DFU bootloader device are available.

miguelwon, 2022-02-16:
Thanks. I know the steps. I use them when uploading to the BlackPill STM32F401C, as well as to the BlackPill STM32F411C when `USB support` is selected to `None`. But for this board (STM32F411C), if I set `USB support` to CDC (because I want to use Serial through USB), I'm not able to upload (with DFU mode).

fpistm, 2022-02-16:
You should check the option bytes setting. Maybe the boot0 pin is not properly configured.