stm32duino/Arduino_Core_STM32 #878: Weird compile error for STM32F410Rx

MCUdude, 2020-01-20:
Hi!
I'm playing around with my Nucleo F410RE board to see if I can get it working with this core.
I've created a generic pinout for it (that I haven't pushed yet), but I'm not able to compile code for it due to a bug (?) in the core system files.

This is the error I get whenever I compile something:
```
In file included from C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper\src\stm32\timer.c:14:
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper\src\stm32\timer.c: In function 'getTimerUpIrq':
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\cores\arduino/stm32/timer.h:54:19: error: 'TIM1_UP_TIM10_IRQn' undeclared (first use in this function); did you mean 'TIM1_BRK_TIM9_IRQn'?
   54 | #define TIM1_IRQn TIM1_UP_TIM10_IRQn
      |                   ^~~~~~~~~~~~~~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper\src\stm32\timer.c:374:16: note: in expansion of macro 'TIM1_IRQn'
  374 |         IRQn = TIM1_IRQn;
      |                ^~~~~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\cores\arduino/stm32/timer.h:54:19: note: each undeclared identifier is reported only once for each function it appears in
   54 | #define TIM1_IRQn TIM1_UP_TIM10_IRQn
      |                   ^~~~~~~~~~~~~~~~~~
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper\src\stm32\timer.c:374:16: note: in expansion of macro 'TIM1_IRQn'
  374 |         IRQn = TIM1_IRQn;
      |                ^~~~~~~~~
```

However: if I change this line in stm32f410rx.h to `TIM1_UP_TIM10_IRQn` it does compile without any errors:
https://github.com/stm32duino/Arduino_Core_STM32/blob/a37f9590e6752076068f893280ee88fe3d456f36/system/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f410rx.h#L115

Any reason why `TIM1_UP_TIM10_IRQn` isn't defined for any of the F410 targets?

fpistm, 2020-01-20:
Hi @MCUdude 
In fact to simplify IRQ handling, Each IRQn is mapped on a generic name TIMx_IRQn.
So for this F410 it seems not needed to map it as it already exist.
They are some exception already present.

The fix have to be made here:
https://github.com/stm32duino/Arduino_Core_STM32/blob/a37f9590e6752076068f893280ee88fe3d456f36/cores/arduino/stm32/timer.h#L54-L55

``` patch
#elif defined(STM32F2xx) || defined(STM32F4xx) || defined(STM32F7xx)
+ #if defined(TIM10_BASE) && !defined(TIM10_IRQn)
#define TIM1_IRQn TIM1_UP_TIM10_IRQn
#define TIM1_IRQHandler TIM1_UP_TIM10_IRQHandler
+ #endif
```

MCUdude, 2020-01-20:
Great. Is this something I can add together with a "Generic F410Rx pinout" PR?

fpistm, 2020-01-20:
Well, first it need to be tested :wink:

MCUdude, 2020-01-20:
Well, sadly it didn't solve the problem 😢 

```c
#elif defined(STM32F3xx) || defined(STM32L4xx) || defined(STM32WBxx)
#define TIM1_IRQn TIM1_UP_TIM16_IRQn
#define TIM1_IRQHandler TIM1_UP_TIM16_IRQHandler
#elif defined(STM32F2xx) || defined(STM32F4xx) || defined(STM32F7xx)
#if defined(TIM10_BASE) && !defined(TIM10_IRQn)
#define TIM1_IRQn TIM1_UP_TIM10_IRQn
#define TIM1_IRQHandler TIM1_UP_TIM10_IRQHandler
#endif
#elif defined(STM32H7xx) || defined(STM32MP1xx)
#define TIM1_IRQn TIM1_UP_IRQn
#define TIM1_IRQHandler TIM1_UP_IRQHandler
#endif

```

```
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper\src\stm32\timer.c: In function 'getTimerUpIrq':
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper\src\stm32\timer.c:374:16: error: 'TIM1_IRQn' undeclared (first use in this function); did you mean 'TIM5_IRQn'?
  374 |         IRQn = TIM1_IRQn;
      |                ^~~~~~~~~
      |                TIM5_IRQn
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper\src\stm32\timer.c:374:16: note: each undeclared identifier is reported only once for each function it appears in
```

fpistm, 2020-01-20:
Right I didn't saw the UP. I've corrected.
Anyway I will have to check if this could be set in the next preprocess as it defines it properly for TIM1_UP.
`#elif defined(STM32H7xx) || defined(STM32MP1xx)`

MCUdude, 2020-01-20:
@fpistm do you want me to create a PR for the Generic F410Rx variant I'm working on (pretty much done with)? This would probably make it a little easier for you to track this issue down 🙂

fpistm, 2020-01-21:
Yes, thanks @MCUdude

PR #882 (2020-01-22): Fix TIM1_IRQn definition when TIM10 is not defined
Fixes #878
commit 0ff0a2a4 (2020-01-22): Fix TIM1_IRQn definition when TIM10 is not defined

Fixes #878
closed 2020-01-22 by fpistm