stm32duino/Arduino_Core_STM32 #879: SoftwareSerial hardware timer interrupt not being called?

Bambofy, 2020-01-20:
**Describe the bug**
When using the SoftwareSerial library to communicate with SIM800L, the STM32 freezes at the print function.

**To Reproduce**
```

// PB7 is connected to the arudino uno's PIN D5
// PB6 is connected to the arduino uno's PIN D6
// See arduino uno sketch below //

// PC4 is connected to SIM800L Tx pin.
// PC5 is connected to the SIM800L Rx pin.

HardwareSerial DebugSerial(PB7, PB6);
SoftwareSerial GSMSerial(PC5, PC4); // see notes posted.

void assert_failed(uint8_t* file, uint32_t line)
{
	DebugSerial.begin(9600);

	// UM1884
	// STM3245re hal guide.pdf
	// page 72/2232
	DebugSerial.print("Wrong parameters values: file ");
	DebugSerial.print((char*)file);
	DebugSerial.print(" on line: ");
	DebugSerial.println(line);
}

void setup()
{
	DebugSerial.begin(9600);
	GSMSerial.begin(9600);

	DebugSerial.println("Started serial debugging.");
}

void loop()
{
	DebugSerial.println("Sent AT command.");
	GSMSerial.print("AT\r\n");

	delay(3000);

	DebugSerial.println("Checking for response...");
	while (GSMSerial.available() > 0)
	{
		char c = GSMSerial.read();
		DebugSerial.print(c);
	}
	DebugSerial.println("- end of response -");
}
```

Arduino uno running this sketch:
```
#include <SoftwareSerial.h>

SoftwareSerial SoftSerial(6, 5);

void setup()
{
  Serial.begin(9600);
  SoftSerial.begin(9600);

  Serial.println("Debugger started");
}

void loop()
{
  if (SoftSerial.available() > 0)
  {
    char c = SoftSerial.read();
    Serial.print(c);
  }
}
```

Steps to reproduce the behavior:
1. Connect an arduino uno to your pc and upload the given sketch.
2. Connect the arduino pins 5 and 6 to stm32l452re pins PB7 and PB6
3. Connect the STM32l542RE pins PC4 and PC5 to a uart device's (sim800l used here) Tx and Rx pins.
4. Build and run the STM32duino sketch.
Result:
- it freezes at the SoftwareSerial GSMSerial(PC5, PC4); function.

Using SoftwareSerial GSMSerial(PC5, PC4); the code freezes.
Using SoftwareSerial GSMSerial(PC4, PC5); the code freezes.

**Expected behavior**
Uno to print data from the GSM to the PC.

 - Name: STM32L542RE

**Additional context**
I investigated the SoftwareSerial() library and the HardwareTimer it uses was not calling the InterruptHandler it uses.

ABOSTM, 2020-01-22:
Hi @Bambofy ,
As far as I understand, you face issue with SoftwareSerial, which is the link between STM32L452RE and GSM SIM800L.
So can you reproduce the issue with only those 2 boards ? I mean without arduino Uno. DebugSerial could then be mapped to Serial so that you output debug to Arduino serial monitor.
Also, you say that STM32 freeze at print. Which print?

Bambofy, 2020-01-22:
@ABOSTM 
I have connected the SIM800L directly to the Uno and the component functions correctly, i can't use HardwareSerial since the USART3 TX is connected to the SIM800L TX unfortunatly, this is why i am trying the SoftwareSerial.

STM32 freezes at the code
"SoftwareSerial GSMSerial(PC5, PC4)"

I have done some debugging and see that it does report the error "Incorrect Pin Value" but then when its configured to the right pins it just freezes up at the declaration of the software serial object. I think there is an issue with the hardware timer and the Rx and Tx pin configuration.
The hardware timer does not call the interrupt function when it should. So i think the Rx Tx pin configuration is breaking the HardwareTimer somehow.

ABOSTM, 2020-01-22:
@Bambofy 
I am totally confused:
My suggestion was to reproduce the problem, with a simplified setup (easier to debug/analyse). So keep STM32 and SIM800 and remove Uno. But you tested with the Uno connected to SIM800.

Also you say 1st that STM32 freezes at print, but now it freezes at `SoftwareSerial GSMSerial(PC5, PC4)`
Does it mean that behavior change in between ? or freeze at print was a misanalysis?

Also you say it reports "Incorrect Pin Value". Did you fix this issue ? how?

Bambofy, 2020-01-22:
Ah, the Uno is used as a debugging tool which re-prints the messages from the STM32 to my pc. Because i cannot connect the STM32 directly via serial port since its a custom pcb.

Yes it freezes at definition in this test sketch, but also freezes at println() when used in my bigger program.

The "incorrect pin value" message was given when i purposely tried the wrong pin numbers but with the correct values, it freezes?

ABOSTM, 2020-01-23:
@Bambofy 
I tested on my side SoftwareSerial on NUCLEO_L452RE, it work perfectly:
I used Example available Arduino example: **SoftwareSerialExample**
I used the same pin as you do (PC4/PC5) on NUCLEO_L452RE
`SoftwareSerial mySerial(PC4, PC5);`

The same example running on another nucleo board and of course I connect both boards with crossing wires (TX connected to RX).

It would be good that you also test each part of your setup solely:
1) Software serial : for example, similarly to what I did, test SoftwareSerial between  L452RE and Uno. without SIM800.
2) SIM800: did you notice that SIM800 power supply range is: 3.4V to 4.4V
So you cannot use 3.3V from MCU power supply. Many people use 3.7V external battery.
3) Also, it would be good to test SIM800 with SoftwareSerial but without connection to Uno (without HardwareSerial). 
Aim is to avoid potential conflict (interrupt preemption, ...) with other part of the system like HardwareSerial.

Bambofy, 2020-01-23:
Hey, thanks for looking into this @ABOSTM!

Attach these pins (even though this configuration is incorrect):
- USART3 Tx (PC4) to SIM800L Tx
- USART3 Rx (PC5) to SIM800L Rx

Then software serial to use the connection:
SoftwareSerial sim800lSerial(PC4, PC5);

This is the same as your setup except the TX is connected to TX and the RX to RX. On my setup this halts the stm32.

I can tried using a power supply and a 3.7V 10000mAH battery but i have tested the power electronics and the SIM800L works fine when i connect to it with the uno.

ABOSTM, 2020-01-24:
@Bambofy,
I am sorry, but I am again confused about the different tests you made and the status.
So let's make it very simple, to verify SoftwareSerial, can you test the following setup which is working on my side:
Remove SIM800, and do not use USART. 
Keep only Uno and STM32L452RE. 
It is mandatory for serial connection to cross TX/RX connections:
Connect Uno pin2 (RX) to STM32 PC5 (TX)
Connect Uno pin3 (TX) to STM32 PC4 (RX)

On both board I used Arduino example 'SoftwareSerialExample', but as you don't have serial on STM32, you need to adapt it. For example you can send constant string from STM32 and once received on Uno, display it on serial monitor. This will test STM32 TX.
Then you can send string from Uno and once received on STM32, blink a led.

On Uno use: 
```
SoftwareSerial mySerial(2, 3); // RX, TX
```
On STM32L452RE use:
```
SoftwareSerial mySerial(PC4, PC5); // RX, TX
```

fpistm, 2020-01-28:
@Bambofy 
Any feedback on this?
Without any feedback it will be closed as it is OK for the Nucleo-L452RE.

Bambofy, 2020-02-06:
Hi, i've been really busy lately but i should have a good test bench setup soon, so i will provide any info i can :)

fpistm, 2020-02-14:
@Bambofy 
I will close it on monday if no real update on this subject.

closed 2020-02-18 by fpistm