stm32duino/Arduino_Core_STM32 #881: STM32CubeIDE to upload code

Mark-81, 2020-01-21:
In the [Getting Started page](https://github.com/stm32duino/wiki/wiki/Getting-Started) I read:

> To upload through SWD (STLink), Serial or DFU, STM32CubeProgrammer needs to be installed. 

I wonder if the new [STM32CubeIDE](https://www.st.com/en/development-tools/stm32cubeide.html) might be used instead.

fpistm, 2020-01-21:
Hi @Mark-81 
The STM32Cube IDE is an IDE not an uploader tool.
AFAIK it embeds the STM32CubeProgrammer.
Looking in the plugins directory of the installation the `STM32_Programmer_CLI.exe` is available: `C:\ST\STM32CubeIDE_1.1.0\STM32CubeIDE\plugins\com.st.stm32cube.ide.mcu.externaltools.cubeprogrammer.win32_1.1.0.201910081157\tools\bin`

If you want use the one installed with it then you simply have to define in your `PATH` environment 
this directory.

fpistm, 2020-01-23:
@Mark-81 
any update on this issue ?

Mark-81, 2020-01-23:
Following your suggestion I added the directory to the path and it works as expected.

closed 2020-01-23 by Mark-81