stm32duino/Arduino_Core_STM32 #886: Handling of VBUS sensing

matthijskooijman, 2020-01-23:
While, reading the code, I was a bit confused by this bit: https://github.com/stm32duino/Arduino_Core_STM32/blob/0ff0a2a45181f9788852c0075bf50c01c6f8fed9/cores/arduino/stm32/usb/usbd_conf.c#L472-L496

In particular, IIUC, it sets `g_hpcd.Init.vbus_sensing_enable` to `ENABLE` when `USE_USB_HS` is defined, but to `DISABLE` when using the OTG_FS peripheral.

Why is this difference there?

It seems it has been there from the start, when HS support was introduced in 861cd071b932a8e46c5561e515785735d3bfe6be, FS had no VBUS sensing and HS was added with VBUS sensing.

Ultimately, I think think this is something that should be board-specific? If the VBUS line is connected to the VBUS pin, then VBUS sensing should be enabled? If not (which is, I think, recommended only for bus-powered devices where VBUS is always present when the MCU is running).

Maybe it would be useful to enable VBUS sensing by default and have a "USB_ENABLE_VBUS_DETECTION" macro in `variant.h` for boards that need it?

fpistm, 2020-01-24:
@matthijskooijman 
When added I'm based on STM32Cube example which I guess enable it.
In most of the case the VBUS is not used.
This is the same for the low power mode which some part of code are available but not used.

fpistm, 2020-01-31:
Interesting article:
https://community.st.com/s/article/FAQ-Management-of-VBUS-sensing-for-USB-device-design

ABOSTM, 2020-02-27:
Hi @matthijskooijman,
I agree with you that our Vbus sensing configuration is not coherent and as said @fpistm it is an heritage of STM32Cube example.
I also agree that it is board dependent because we need the VBUS line is connected to the VBUS pin.
Nevertheless I dig into this feature and I have several remarks:
* Even on the same chip (ex: F746) the sensing pin is different depending on Full Speed PA9 or High Speed PB13
* In High Speed this feature is applicable only when HS connector is degraded to Full Speed switch USE_USB_HS_IN_FS (Sensing from MCU is not applicable to real High Speed)
* In Device Mode: according to USB specifications Vbus sensing is mandatory only for self-powered board (not powered by USB Vbus itself)
  see article mentioned by @fpistm
  But user have the choice either to enable this hardware feature, using the dedicated Vbus pin,
  or to use any pin with software management of the sensing. 
* According to our USB expert, it is not a big deal if sensing is not enabled, even in case of self-powered board.
  It is certainly a specification violation but it doesn't prevent usb to work properly and there is no risk.
  (this is the reason why STM32Cube examples doesn't really care about this setting)
* STM32Nuleo144 boards and most Discovery boards (maybe all) have PA9 connected to Vbus but not PB13.
* Most STM32 boards have a jumper to select MCU supply (USB Vbus, ST link, ...).

Taking all those remarks into consideration, it looks very complex to distinguish FS/HS/HS_IN_FS, impossible to detect Jumper configuration... 
And no added value: in any case USB is functional.
So we propose with @fpistm to disable this feature by default.
and as you suggested, add a switch to enable it when user require it (USBD_VBUS_DETECTION_ENABLE).
So it is up to user to enable the feature and make sure of Vbus pin connection.
I will submit a PR.

matthijskooijman, 2020-02-28:
Thanks for the detailed analysis. I had not realized that there could be different pins in use, which indeed complicates things. Regardless, it really depends on the board whether VBUS was connected and thus whether VBUS sensing should be enabled or not, so your proposal makes complete sense to me.

The idea now is that different boards can set `USBD_VBUS_DETECTION_ENABLE` to let VBUS be enabled? If so, should this macro not be set by the boards that do have `VBUS` connected? That would make VBUS detection work as expected, regardless of how the board is powered *and* serves as an example for new boards at the same time? Maybe also add this to the template variant?

Or do all currently supported boards have the FS VBUS detection connected but not HS (or vice versa)? That does makes me wonder: Should a board (variant) be able to *separately* enable VBUS detection for FS/HS? That might better reflect the hardware setup. OTOH, I guess a variant could also check things like `USE_USB_HS_IN_FS` and set `USBD_VBUS_DETECTION_ENABLE` accordingly.

ABOSTM, 2020-02-28:
The idea was to do nothing else. Let user activate it only when he required it.
But from my point of view, it is useless to spent much more time on this, as there is no added value except if your intend to pass certification on your product.
Nevertheless adding something in the template is a good idea. Done.

matthijskooijman, 2020-02-28:
Agreed, thanks!

closed 2020-02-28 by fpistm

beachmiles, 2024-01-30:
Ideally there would be a drop down asking if device is self powered or USB bus powered and then select the correct option for Activate_VBUS.

fpistm, 2024-01-31:
> Ideally there would be a drop down asking if device is self powered or USB bus powered and then select the correct option for Activate_VBUS.

Do not hesitate to provide a PR then we could discuss / review your solution.

PR #954 (2020-02-28): USB: Disable Vbus sensing by default
**Summary**
USB: Disable Vbus sensing by default

Tested on STM32F746-DISCO: 
Full Speed and High Speed
ST-link power supply and USB Power supply 

Fixes #886
commit 4aafc222 (2020-02-28): USB: Disable Vbus sensing by default

Fixes #886