stm32duino/Arduino_Core_STM32 #894: Use this repo directly with platformio

ramboerik, 2020-01-28:
**Is your feature request/improvement related to a problem? Please describe.**
Currently it is **very** hard to run this repo directly with platformio. The only way to use it with platformio is to use their [pre-packed](http://dl.platformio.org…32-4.10700.200103.tar.gz) versions that also include some build files and the CMSIS/HAL dependency. Other platforms, such as [esp32](https://github.com/espressif/arduino-esp32), includes all required files for platformio and it's possible to directly from platformio point to the latest dev branch of their repo. It would be very useful if the same thing was possible for this repo. I think this has the potential to allow more people to use and contribute to the repo.

What I would like to specify in platformio's config file:
```
platform_packages = framework-arduinoststm32@https://github.com/stm32duino/Arduino_Core_STM32.git
```

**Describe the solution you'd like**
Include necessary build files for platformio in the repo. Some of them are:
* package.json
* tools/
* Find a solution to handle CMSIS/HAL dependency.

fpistm, 2020-01-28:
Hi @ramboerik 
We support only Arduino IDE and I'm not use PIO.
If only a package.json is needed (and it does not break Arduino IDE usage), you can submit a PR.

ramboerik, 2020-01-28:
I'm pretty sure package.json and the few build files in tools don't break arduino IDE, the problem is probably how to handle the CMSIS/HAL dependencies. Where can I find the repos of these and what versions that head of this repo depends upon?

fpistm, 2020-01-28:
I think also the JSON will not break.
CMSIS is an external package:
https://github.com/stm32duino/ArduinoModule-CMSIS
Tools are available here:
https://github.com/stm32duino/Arduino_Tools
Toolchains here:
https://github.com/xpack-dev-tools/arm-none-eabi-gcc-xpack/releases/tag/v9.2.1-1.1

If I well understand,  you want include the tools in this repo? If true, then forget that, tools are in an independent package as required by Arduino.

ramboerik, 2020-01-28:
Thanks, No I don't want to merge the repos but maybe add pio's build script and modify it to download/checkout CMSIS as an external dependency. I don't really know how to handle it in pio but I understand that it isn't an option to create some sort of all in one repo :smile:

fpistm, 2020-01-28:
OK.
I guess you should ask to PIO some support about that. Maybe in your issue to update the PIO package?
https://github.com/platformio/platform-ststm32/issues/345

ramboerik, 2020-01-28:
Hehe, yes. I tried to reach out the them as well! This question might be more related to their work than yours. I will keep on investigating how to proceed. I guess this issue can be closed.

fpistm, 2020-01-30:
I close it.
If any action required here for PIO support. Do not hesitate to provide a PR.

closed 2020-01-30 by fpistm