stm32duino/Arduino_Core_STM32 #895: SoftwareSerial not working on rel 1.8.0

pacman1945, 2020-01-29:
I'm using SoftwareSerial on Bluepill (STM32F103C8) & STM32F031K6.  Both boards are working fine under Core rel 1.7.0.  After upgrading to rel 1.8.0, SoftwareSerial stopped working on these 2 boards.

The following simple sketch can demonstrate the SoftwareSerial problem on rel 1.8.0.  The same sketch worked fine on rel 1.7.0.

```
#include <SoftwareSerial.h>

SoftwareSerial mySerial(PA10, PA9); // RX, TX

void setup() {
  mySerial.begin(38400);
  mySerial.println("Hello world");
}

void loop() { // run over and over
    mySerial.println("11223344556677889900  11223344556677889900  11223344556677889900");
    delay(1000);
}
```

stas2z, 2020-01-29:
I can assume it's related to #841

pacman1945, 2020-01-30:
@stas2z, thanks for the link.  This is indeed the same problem.  I've tested the patch and it's working fine.

closed 2020-01-30 by pacman1945