stm32duino/Arduino_Core_STM32 #896: DAC on generic F103RC

ARKbr, 2020-01-29:
I cant find any exemple on how use the DAC, can someone give me a light?

I can just use the AnalogWrite() function on the specified pins, or i need to configure any other options?

Thanks in advance :)

fpistm, 2020-01-30:
@ARKbr 

First of all, please use the forum (https://www.stm32duino.com) for this kind of question as described in the [README.md](https://github.com/stm32duino/Arduino_Core_STM32#troubleshooting):

> For question, support, ...,  you could submit a topic on the [stm32duino forum](http://stm32duino.com).

and also in the GitHub issue template.

For the DAC, yes. Simply use the `analogWrite()` on a pin with DAC capabilities.

closed 2020-01-30 by fpistm

ARKbr, 2020-01-30:
@fpistm Sorry for the incovenience and thanks a lot for the help :)

I hadn't seen that the forum was back online, other times that i tried to enter it was offline