stm32duino/Arduino_Core_STM32 #907: Cannot compile for STM32F410Cx targets

MCUdude, 2020-02-03:
There are a few STM32 targets I still would like to add to this project. One of them is the STM32F410Cx series.

I do have a working variants file for them. However, I'm not able to compile for either of them (STM32F410C8 or STM32F410CB). Here's the error I'm getting:

```
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper\src\stm32\low_power.c: In function 'LowPower_EnableWakeUpPin':
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper\src\stm32\low_power.c:124:7: error: duplicate case value
  124 |       case SYS_WKUP3 :
      |       ^~~~
C:\Users\user\AppData\Local\Arduino15\packages\STM32\hardware\stm32\1.8.0\libraries\SrcWrapper\src\stm32\low_power.c:114:7: note: previously used here
  114 |       case SYS_WKUP2 :
      |       ^~~~
```

This is the added boards.txt entry:
```
# Generic F410CB
GenF4.menu.pnum.Generic_F410CB=Generic F410CB
GenF4.menu.pnum.Generic_F410CB.upload.maximum_size=131072
GenF4.menu.pnum.Generic_F410CB.upload.maximum_data_size=32768
GenF4.menu.pnum.Generic_F410CB.build.board=GENERIC_F410CB
GenF4.menu.pnum.Generic_F410CB.build.product_line=STM32F410Cx
GenF4.menu.pnum.Generic_F410CB.build.variant=Generic_F410Cx

# Generic F410C8
GenF4.menu.pnum.Generic_F410C8=Generic F410C8
GenF4.menu.pnum.Generic_F410C8.upload.maximum_size=65536
GenF4.menu.pnum.Generic_F410C8.upload.maximum_data_size=32768
GenF4.menu.pnum.Generic_F410C8.build.board=GENERIC_F410C8
GenF4.menu.pnum.Generic_F410C8.build.product_line=STM32F410Cx
GenF4.menu.pnum.Generic_F410C8.build.variant=Generic_F410Cx
```

And attached you can find the variants file. once this is sorted out I can submit PR for this variant.
[Generic_F410Cx.zip](https://github.com/stm32duino/Arduino_Core_STM32/files/4147876/Generic_F410Cx.zip)

fpistm, 2020-02-03:
@MCUdude 
This is due to missing `SYS_WKUPx` definition in the `PinNamesVar.h`.
This file is generated thanks CubeMx XML description file anyway sometimes some WKUP pins are missing which leads to this issue as the one not found exists and defined to NC.
In that case, you have to check in the MCU datasheet which one is missing and define it manually.

fpistm, 2020-02-03:
https://github.com/stm32duino/Arduino_Tools/blob/3cf9602fa4c0c86c2ff80eaf0e2bfd4a64de9f92/src/genpinmap/Arduino/STM32F410C(8-B)Tx/PinNamesVar.h#L6
Should be:
```
- SYS_WKUP2 = NC,
+ SYS_WKUP2 = PC_0,
```
https://github.com/stm32duino/Arduino_Tools/blob/3cf9602fa4c0c86c2ff80eaf0e2bfd4a64de9f92/src/genpinmap/Arduino/STM32F410C(8-B)Tx/PinNamesVar.h#L9
Should be:
```
- SYS_WKUP3 = NC,
+ SYS_WKUP3 = PC_1,
```

MCUdude, 2020-02-03:
Thanks, that did the trick! You can expect a PR soon 🙂

closed 2020-02-03 by MCUdude