stm32duino/Arduino_Core_STM32 #917: can't entrance HardwareTimer callback function

zhaorui4142, 2020-02-05:
when i use stm32f103c8t6(blue pill) runing a example(Timebase_callback) ,find Timer dosen't work(counter not running),then i open "xx\Arduino15\packages\STM32\hardware\stm32\1.8.0\system\Drivers\STM32F1xx_HAL_Driver\Src\HardwareTimer.cpp" in line 155 and modified as flow:
void HardwareTimer::resume(void)
{
  // Clear flag and ennable IT
  if (callbacks[0] != NULL) {
    __HAL_TIM_CLEAR_FLAG(&(_timerObj.handle), TIM_FLAG_UPDATE);
    __HAL_TIM_ENABLE_IT(&(_timerObj.handle), TIM_IT_UPDATE);
  }
  // Resume all channels
  resumeChannel(1);
  resumeChannel(2);
  resumeChannel(3);
  resumeChannel(4);
  **__HAL_TIM_ENABLE(&(_timerObj.handle));**//add this line,to start timer
}
rebuilt and upload, then timer is start worked，and succesfull entranced HardTimer callback function.

fpistm, 2020-02-05:
Hi @zhaorui4142 ,
I guess you used the master of the  STM32Example library which is not released yet.
Use the example of the last release then it will work.
There is a change : 
https://github.com/stm32duino/STM32Examples/commit/60f8b8cbbea7bd4ef0beee7c275ce60a3352f716
`MyTim->setMode` is needed with 1.8.0.

closed 2020-02-05 by fpistm

wuzhongjie1997, 2020-02-20:
Does this mean I have to initialize a channel to some mode even I am not using any channel? e.g. if I only want the timer to perform regular callbacks? I do not want the timer to mess up with my pin configuration...

ABOSTM, 2020-02-20:
Hi @wuzhongjie1997 ,
If you are on 1.8.0 version of the core, then yes you need the 'old' way `setMode(Output1_channel, TIMER_OUTPUT_COMPARE, NC);` even if you don't use the channel. Thanks to `NC` pin parameter it will not mess up with your pin configuration.
This constraint has been removed, but it is not yet officially released.
So, for future release, or if you use the master branch of github, this is no more necessary you can get ride of `setMode()` for this use case.

wuzhongjie1997, 2020-02-20:
Thanks, I simply modified the library and recompiled. It worked. (Many thanks for your great work!)