stm32duino/Arduino_Core_STM32 #919: SoftwareSerial hardware timer interrupt  not being called?

lab-mathias-claussen, 2020-02-06:
STM32F072C8T6 / Elektor LoRa Node 

Seems like #879 the interrupt fir the Hardware timer is not getting called. We can see that the IO Pins get configured ( PA2 and PA3 ) in this case. 

To Reproduce:

Fit a STM32F072C8T6 on a bluepill and put the following code using version 1.8 of the core on the board:

> /* Includes */
> #include <hal/hal.h>
> #include <SPI.h>
> #include <Wire.h>
> 
> #include <SoftwareSerial.h>
> #include <EEPROM.h>
> 
> SoftwareSerial SerialX(PA3, PA2); //RX, TX
> /**************************************************************************************************
>  *    Function      : print_array
>  *    Description   : will print an array of data as hex
>  *    Input         : uint8_t * data , uint32_t len 
>  *    Output        : none
>  *    Remarks       : none
>  **************************************************************************************************/
> void print_array(uint8_t * data , uint32_t len ){
> 
>   for(uint32_t i=0;i<len;i++){
>            Serial.print("0x");
>            Serial.print(*data++);
>            Serial.print(" ");
>   }
>   Serial.println("");
> } 
> 
> 
> void setup() {
>   Serial.begin(115200);
>   SerialX.begin(9600);
>   Serial.println("Boot");
>    
> }
> 
> void loop() {
>   SerialX.write(0xAA);
>   while(SerialX.available() ){
>     Serial.write(SerialX.read());
>   }
>   Serial.write(0x55);
>   
> }

What will be shown on USART 1 is:

> Boot
> U

And at that Point the code seems to stop at all
[image]

PA2 is TX and PA3 is RX for hardware serial output we use PA9 / PA10 

Currently we try to investigate this further but stummbeled also accross some other issues that may lay down int the variant itself. The code works fine on a STM32F103 blue pill

stas2z, 2020-02-06:
#895

fpistm, 2020-02-06:
Hi @lab-mathias-claussen 

You can check the issue provided by  @stas2z

Anyway about this:

> Fit a STM32F072C8T6 on a bluepill

What would you mean by this ? You replaced the STM32F103 by a STM32F072C8T6  ?
So this is a custom board?
Which variant you used in this case?

lab-mathias-claussen, 2020-02-06:
It uses the ELEKTOR_F072C8 variant, an we replaced the STM32F103 by a STM32F072C8T6. This is kind of a custom board for internal development to mimic the ELEKTOR_F072C8 variant. 

[image]

lab-mathias-claussen, 2020-02-06:
Fixed with current code from Git repository
Thanks

closed 2020-02-06 by lab-mathias-claussen