stm32duino/Arduino_Core_STM32 #927: [Question]disable the JTAG function on the F405 board.

nopnop2002, 2020-02-08:
I want to disable the JTAG function on the F405 board.
The following APIs were available on the F103 board:
```
  pinF1_DisconnectDebug(PB_3); // disable Serial wire JTAG configuration
  pinF1_DisconnectDebug(PB_4); // disable Serial wire JTAG configuration
  pinF1_DisconnectDebug(PA_15); // disable Serial wire JTAG configuration
```

I can't find an API with the same function on the F405 board.

stas2z, 2020-02-08:
AFAIK f4 doesn't have special remap register to disable jtag like f1, just use pinMode on jtag pin to set common mode and it's jtag function will be disabled

fpistm, 2020-02-08:
Hi @nopnop2002 
First of all, please use the forum for this kind of request. Primary support is done on it.

@stas2z is right. It is linked to AF (Alternate function) of a GPIO.
So use it as an other.
https://github.com/stm32duino/Arduino_Core_STM32/blob/e13d70e09cb79b2aadc8ed4799e7d198a9440be6/system/Drivers/STM32F4xx_HAL_Driver/Inc/stm32f4xx_hal_gpio_ex.h#L397

closed 2020-02-08 by fpistm