stm32duino/Arduino_Core_STM32 #929: Problem in building Arduino_Core_STM32 based project in Eclipse C/C++ IDE

engrziarehman, 2020-02-10:
I installed the fresh eclipse C/C++ IDE 2019.
In "Preferences->arduino" I add the link for stm32 board (https://github.com/stm32duino/BoardManagerFiles/raw/master/stm32/package_stm_index.json).
In "Help"-> Arduino Download Manager, installed STM32 core.
Then i selected bluepill as New Launch Target, All of the stm32 list of MCUs appeared fine.
but when i click build the following error appears (attached).

> Undefined key build.source.path

[image]

VitorBoss, 2020-02-10:
Did you follow those steps?
https://gnu-mcu-eclipse.github.io/plugins/packs-manager/

https://gnu-mcu-eclipse.github.io/tutorials/blinky-arm/

engrziarehman, 2020-02-10:
@VitorBoss Thankyou for your reply, I have not followed this step, when i installed STM32 core inside download manager, it downloaded all of the pre-requisites. do you still feel i need to follow above steps for arduino environment?.
further i brief, when i select AVR based core, eclipse build fine and locate Arduino.h included file correctly but when i select this Arduino_Core_STM32 you can see in image, eclipse can't even locate `Arduino.h` file,

fpistm, 2020-02-10:
@engrziarehman 
Unfortunately, we doesn't support another IDE than Arduino.
You can try to open a topic about that on the forum to get help.

closed 2020-02-10 by fpistm

jamesjian007, 2020-02-18:
Thanks. Now make sense.