stm32duino/Arduino_Core_STM32 #930: EEPROM usage hangs when used with RTC

jgromes, 2020-02-10:
Hi, I'm trying to use this library together with EEPROM, but for some reason, calls to EEPROM methods (put/get) seem to hang when using LowPower - the following sketch hangs at `EEPROM.put()` and never reaches `EEPROM.get()` or another loop.

Board: Nucleo L452RE-P
STM32 core version: cloned from repo
Sketch to reproduce:
```c++
#include <STM32LowPower.h>
#include <EEPROM.h>

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);
  Serial.begin(115200);
  LowPower.begin();
}

void loop() {
  // print stuff to serial port
  Serial.println("loop");
  delay(100);

  // sleep twice
  digitalWrite(LED_BUILTIN, HIGH);
  LowPower.deepSleep(1000);
  digitalWrite(LED_BUILTIN, LOW);
  LowPower.deepSleep(1000);

  // EEPROM put - hangs here
  Serial.println("eeprom put");
  delay(100);
  EEPROM.put(0, 0xFF);

  // EEPROM get
  Serial.println("eeprom get");
  delay(100);
  uint8_t val = 0;
  EEPROM.get(0, val);
  Serial.println(val);
}
```

Thanks!

ABOSTM, 2020-02-10:
Hi @jgromes ,
Problem is not linked to RTC, just a combination of circumstances:
Sometimes, `eeprom_buffer[]  ` address was aligned on 2 bytes (16 bits),
but due to cast on (uint64_t *) compiler uses a asm instruction LDRD to read Double word.
This instruction needs some 4bytes/8bytes alignment (depending on core/architecture used).
Misalignment cause hardfault.
It is now fixed

jgromes, 2020-02-10:
That's absolutely brilliant, thank you so much!

closed 2020-02-11 by fpistm

HamidSaffari, 2020-04-20:
Hi, creators.
The same problem happens to me as I described here: https://github.com/stm32duino/Arduino_Core_STM32/issues/1038
although I used this fix: https://github.com/ABOSTM/Arduino_Core_STM32/commit/b8d7c44f2a66a08684f8cf250a27743bab6e34d3

HappySamuel, 2021-07-30:
Hi 

Is that mean EEPROM.write() from <EEPROM.h> is safe to use now? Does this apply the same to using on STM32F103Cx series?

Best,
Samuel

ABOSTM, 2021-07-30:
Hi @HappySamuel,
Yes, bug is now fixed.
Even for STM32F103Cx

HappySamuel, 2021-08-24:
Hi @ABOSTM , @fpistm 

Is there a way to check how many times left for able to do EEPROM write for the STM32F103Cx series left??

Best,
Samuel

fpistm, 2021-08-24:
Hi @HappySamuel 
Unfortunately not. If you really need access several time to EEPROM, the best way is to use an I2C EEPROM, I guess.
Here the characteristic of the STM32F103Cx Flash:
[image]

HappySamuel, 2021-08-25:
Hi @fpistm 

Any suggested examples or libraries to use I2C EEPROM?

Best,
Samuel

fpistm, 2021-08-25:
> 
> 
> Hi @fpistm
> 
> Any suggested examples or libraries to use I2C EEPROM?
> 
> Best,
> Samuel

https://www.arduino.cc/reference/en/libraries/exteeprom/

HappySamuel, 2021-08-25:
Hi @fpistm 

Thank you for the sharing. Will test it out when i have an I2C EEPROM.

Best,
Samuel

PR #931 (2020-02-11): EEPROM hardfault: eeprom_buffer should be 8 bytes aligned
**Summary**
EEPROM hardfault: eeprom_buffer should be 8 bytes aligned

Due to some cast on (uint64_t *) of eeprom_buffer,
compiler may use asm instruction LDRD which required an alignment:
4 bytes on Armv7 architecture,
and 8 bytes (depending on SCTLR.U) on Armv6 architecture.
To fit to all cases, alignment is made on 8 bytes.

Fixes #930
commit c29e234d (2020-02-11): Fix EEPROM hardfault: eeprom_buffer should be 8 bytes aligned (#931)

Due to some cast on (uint64_t *) of eeprom_buffer,
compiler may use asm instruction LDRD which required an alignment:
4 bytes on Armv7 architecture,
and 8 bytes (depending on SCTLR.U) on Armv6 architecture.
To fit to all cases, alignment is made on 8 bytes.

Fixes #930