stm32duino/Arduino_Core_STM32 #932: NUCLEO_L452RE-P: wrong LED_BUILTIN pin association?

Alteregoxxx, 2020-02-10:
Hello,
in the the variant.h file of the board NUCLEO_L452RE, it seems there is a wrong pin assigned to the builtin LD4 USER (green led) in case one is using a NUCLEO_L452RE-P variant.

Actually, if I'm not understanding it wrong, the LED_BUILTIN, at line 156, is always assigned to physical pin 13:

// On-board LED pin number
#define LED_BUILTIN             13

while, if I'm not wrong, the correct assignment should be:
// On-board LED pin number
#define LED_BUILTIN             PB13

Am I wrong?

**Desktop:**
 - OS: Windows 10
 - STM32 core version: 1.8.0

**Board:**
 - Name: NUCLEO_L452RE-P

fpistm, 2020-02-10:
Hi @Alteregoxxx 
This is correct:
https://github.com/stm32duino/Arduino_Core_STM32/blob/97e3f523da328b7a2ac435dcc26f24099af19914/variants/NUCLEO_L452RE/variant.h#L100

13 is the pin number (index) and point to the correct pin name.

Alteregoxxx, 2020-02-11:
You're right. I was just exploring the way Arduino manage the "pins things".
I was not aware that the assignment to the physical pin has to happen trough the array `const PinName digitalPin[]`; hence 13 stands for the 13th element of that array that, for the board L452RE-P is, in fact, the physical pin PB_13, while for the board L452RE is the physical pin PA_5.

Thanks for the clarification and sorry for having bothered you!

closed 2020-02-11 by Alteregoxxx

fpistm, 2020-02-11:
No worries.
Welcome 😉