stm32duino/Arduino_Core_STM32 #934: noInterrupts problem

ricaun, 2020-02-12:
**Describe the bug**

Hello, I'm trying to use `noInterrupts()` on my library and I notice the board stop working for no reason.
I create a simple sketch and the board stops when I call the `noInterrupts()`. I tried to call `interrupts()` first but nothing.
Does anyone have this problem?

**To Reproduce**

```
void setup() {
  Serial.begin(115200);
  Serial.println("Start noInterrupt test!");
  delay(100);
  noInterrupts(); // <------------- this makes the board to block and never goes to the loop
}

void loop() {
  Serial.println("noInterrupt loop");
  delay(100);
}

```

**Desktop**
 - OS: Windows
 - Arduino IDE version: 1.8.8
 - STM32 core version: 1.8.0
 - Upload method: STLink

**Board**
 - Name: Generic STM32F1 series - BluePill F103C8

**Additional context**

I tested with other boards and has the same problem.

AnHardt, 2020-02-12:
"Serial.println()" needs the interrupts to be **on** - else it can't print anything.

ricaun, 2020-02-12:
> 
> 
> "Serial.println()" needs the interrupts to be **on** - else it can't print anything.
> Same for delay() - the clock delay() is looking for ist just not running.

This is interesting, the `noInterrupts()` disable the `HAL_GetTick`. `millis()` and `delay()` stop working...

The `delay()` stays on a while forever. `delayMicroseconds` does not use any `HAL_GetTick` relation, and does not block.

```
void setup() {
  Serial.begin(115200);
  Serial.println("Start noInterrupt test!");
}

void loop() {
  noInterrupts();
  Serial.println("noInterrupt loop");
  delayMicroseconds(100000); // wait 100ms
  interrupts();
  delay(1000);
}
```

This code above works!

fpistm, 2020-02-12:
Hi @ricaun 
@AnHardt is right.
For this kind of question, please use the primary support forum: stm32duino.com

closed 2020-02-12 by fpistm

AnHardt, 2020-02-12:
The serial.println() works now, because when sent during the interrupts are off, the print only fills the output-buffer. When the interrupts are back again the buffer can be sent.

ricaun, 2020-03-27:
I kinda found a-ok solution, to not block the delay forever when the interrupt is disabled.

```
uint32_t HAL_GetTick(void)
{
  extern __IO uint32_t uwTick;

  /* Read PRIMASK register, check interrupt status before disable them */
  /* Returns 0 if they are enabled, or non-zero if disabled */
  if (__get_PRIMASK())
  {
    if (SCB->ICSR & SCB_ICSR_PENDSTSET_Msk)
    {
      SCB->ICSR = SCB_ICSR_PENDSTCLR_Msk;
      HAL_IncTick();
    }
  }

  return uwTick;
```

This code rewrite the `HAL_GetTick` function and add an interrupt check and if the interrupt is disabled the code add some `HAL_IncTick`.