stm32duino/Arduino_Core_STM32 #939: Calling analogWrite multiple times causes glitches in PWM signal

ktand, 2020-02-16:
**Describe the bug**
Calling analogWrite multiple times causes glitches in PWM signal

**To Reproduce**
#include <Arduino.h>

void setup() {
  pinMode(PC_6, OUTPUT);
}

void loop() {
  analogWrite(PC_6, 128);
  delayMicroseconds(50);
}

**Expected behavior**
The PWM signal should be glitch-free if analogWrite is called repeatedly with the same value. When PWM has been set up for the selected pin the next call to analogWrite should just update the PWM duty cycle. Now it appears that it restarts the PWM signal.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - PlatformIO: ST STM32 6.0.0
 - STM32 core version: 1.7.0

**Board (please complete the following information):**
 - Name: armed_v1

**Additional context**
I'm using STM32 core version 1.7.0 as a part of PlatformIO-ststm32 6.0.0. I have checked the code and the behaviour should be the same in 1.8.0.

ABOSTM, 2020-02-17:
Hi @ktand ,

First thee are some issues in your sketch:
1) `pinMode(PC_6, OUTPUT);`
this function call is useless because `analogWrite()` will do the GPIO configuration
2) `pinMode()` and `analogWrite()` API accept pin number (PC6) but not pin name (PC_6)

Then, let's discuss the glitch:
with  `analogWrite()` there is a complete reconfiguration of GPIO and HardwareTimer, thus there are intermediate steps where output is deactivated (corresponding to the glitch) and so, with this API, glitch is inevitable.

It is still possible to get ride of glitch by controlling directly HardwareTimer (see below).
For details about Hardware timer:
* See wiki: https://github.com/stm32duino/wiki/wiki/HardwareTimer-library
* See examples of HardwareTimer usage:
https://github.com/stm32duino/STM32Examples/tree/master/examples/Peripherals/HardwareTimer
```
#include <Arduino.h>

#define pin PC6

uint32_t channel;
HardwareTimer *MyTim;

void setup() {
  TIM_TypeDef *Instance = (TIM_TypeDef *)pinmap_peripheral(digitalPinToPinName(pin), PinMap_PWM);
  channel = STM_PIN_CHANNEL(pinmap_function(digitalPinToPinName(pin), PinMap_PWM));

  MyTim = new HardwareTimer(Instance);
  MyTim->setPWM(channel, pin, 10000, 50); // 10k Hz 50% dutycycle
}

void loop() {
  MyTim->setCaptureCompare(channel, 50, PERCENT_COMPARE_FORMAT);
  delayMicroseconds(50);
}
```

ktand, 2020-02-17:
Regarding PC_6 vs PC6, my mistake when I created the sample, sorry.

Regarding the glitch. If PWM is already started on the pin, why can't the core just reconfigure the duty cycle? I'm sure the workaround will be fine (thanks for the example) but in this case the analogWrite is called by code supporting many platforms (Marlin 3D printer F/W).

ABOSTM, 2020-02-17:
Hi @ktand,
I understand your point and I propose a PR to provide a new API and use it in case of analogWrite.
See: https://github.com/stm32duino/Arduino_Core_STM32/pull/941

PR #941 (2020-02-17): Hardwaretimer analogWrite() no glitch
**Summary**

* HardwareTimer: Add API to get configured mode
* analogWrite: Configure HardwareTimer mode only if not previously done  
This avoid glitch when calling analoWrite multiple time.
Also don't start again timer otherwise ther is also some glitches

Fixes #939
commit bdcc98ed (2020-02-17): analogWrite: Configure HardwareTimer mode only if not previously done

This avoid glitch when calling analoWrite multiple time.
Also don't start again timer otherwise ther is also some glitches

Fixes #939
closed 2020-02-17 by fpistm