stm32duino/Arduino_Core_STM32 #943: Timebase_callback:  error: 'TIM_TypeDef' was not declared in this scope

SalimF, 2020-02-18:
Hi I'm trying to get Timer of stm32F103 board works on arduino , the works perfect with basic example and I successfully upload to it using usb cable 
But when I tray this example sketch  [Timebase_callback.ino](https://github.com/stm32duino/STM32Examples/blob/master/examples/Peripherals/HardwareTimer/Timebase_callback/Timebase_callback.ino)
I git this error message :- 
```
/mnt/disk2/Video/STM32_Cortex-M3/EXAMPLES/Timebase_callback/Timebase_callback.ino:
 In function 'void setup()':
Timebase_callback:25:3: error: 'TIM_TypeDef' was not declared in this scope
   TIM_TypeDef *Instance = TIM2;
   ^
```
  I even try on of the example code from the Wiki 
` HardwareTimer *MyTim = new HardwareTimer(TIM3); `
and got same error 
looks none of STM32 timer function works fro me .

**Desktop (please complete the following information):**
 - OS: [Linux mint 18.04]
 - Arduino IDE version: [e.g. 1.8.9]
 - STM32 core version: [e.g.  1.5.0]
 - Tools menu settings if not the : [default]
 - Upload method: [ usb]
screenshot of arduino IDE settings :- 
https://i.ibb.co/Wfg5xCy/Arduino-IDE-01-55-35.png
 
**Board (please complete the following information):**
 - Name: [BlackPill F103C(8-B)]
 - Hardware Revision: [ ] 
**Additional context**

stas2z, 2020-02-18:
As i can see on your screenshot, you have rogers's core installed, not this one

https://github.com/rogerclarkmelbourne/Arduino_STM32

That core doesnt have TIM_TypeDef definition, cuz its not a HAL based

fpistm, 2020-02-18:
Hi @SalimF 
As @stas2z said those examples are for this core and you use them with Roger's core.

closed 2020-02-18 by fpistm

SalimF, 2020-02-18:
> As i can see on your screenshot, you have roger's core installed, not this one
> 

thanks for reply honesty  i tried many bootloader and I can't remember which is last one I already put [Arduino_Core_STM32](https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json) on boards mangers so I add roger's core too , when I hookup the usb cable it show me this info
```
BN: Maple Mini
VID: 0x1EAF
PID: 0x0004
```
anyway I'll try to re-flash Arduino_Core_STM32
Alright even I post reply everything works fine now I upload the sketch with no errors , Thanks @stas2z & @fpistm .
https://i.ibb.co/3TK4KzP/ARDUIONO-IDE-BESTsetting-13-24-00.png