stm32duino/Arduino_Core_STM32 #944: PWM with external sync

uzi18, 2020-02-18:
Hi
I need some advice, have external 100Hz sync signal and want to control external gpio with pwm signal.
Is it possible to have it full hardware solution with timer in use?
Have cpu F407ZET boards for tests.

@ABOSTM as you are timer guru maybe you can guide me in this matter?

ABOSTM, 2020-02-18:
Hi @uzi18 ,
1st it should be better to raise such question on the forum: https://www.stm32duino.com/
Then I don't really understand what you want to do: 
generate a 100 Hz signal with a PWM ? or to synchronize an external GPIO with  an external 100Hz signal?

uzi18, 2020-02-18:
ok my fault, want to have internal gpio driven with pwm (period 10uS), but synced from external 100Hz signal - eg. on rising edge of sync signal want to have start counting  of pwm period

did not know about forum is working

stas2z, 2020-02-18:
@uzi18 Its possible with HAL, google for timer ETR and slave mode

uzi18, 2020-02-18:
@stas2z thx will dig there :)

ABOSTM, 2020-02-18:
@uzi18 , If you really need full hardware solution, you will need HAL (as said @stas2z ) 
Otherwise, with Arduino API you can have a partial hardware solution, for example do such use case:
* Configure a HardwareTimer for 10us  PWM
* Configure an interrupt on external GPIO edge.
* Within corresponding callback,  reset the HardwarteTimer counter (`setCount()`) and start HardwareTimer.

stas2z, 2020-02-18:
> @uzi18 , If you really need full hardware solution, you will need HAL (as said @stas2z )
> Otherwise, with Arduino API you can have a partial hardware solution, for example do such use case:
>
> * [...]

I assume just toggling gpio pin in exti callback will be faster than extra timer stuff )

stas2z, 2020-02-18:
Sorry, ive not read properly what exactly needed, so 8ve deleted my stupid answer) @ABOSTM is right

closed 2020-02-19 by fpistm

uzi18, 2020-02-19:
@ABOSTM have this implemented like that in AVR8, but looks like F4 hardware is more cappable to do it in hardware - need to read abou advanced timers/counters.
Maybe will use registers directly - without HAL.
Thanks for hints.

uzi18, 2020-02-20:
It looks like one shoot mode will do the job here

ABOSTM, 2020-02-20:
I don't know what you call one shoot mode. Is it One Pulse mode ? 
If yes  it is not supported by Arduino API (but supported by HAL). And it means also that I didn't understand what you really want to do. Only a single pulse of 10us synced with external edge?

AnHardt, 2020-02-20:
A common use case for this would be a AC (Main power) dimmer. 
1 Measure the number of ticks between voltage crosses zero (M). Calculate N from M for the PWM- ratio.
2 Detect crossing zero. Start the timer.
3 After N ticks switch a pin on to start the triac.
4 After M minus a bit switch the pin off to stop the triac at next zero crossing.
5 goto 2

fpistm, 2020-02-20:
Please guy,
use the forum to discuss 😉 
This could probably help and/or be informative for other.
Thanks

uzi18, 2020-02-20:
@AnHardt this could be automatic with one pulse mode, you only change settings
@fpistm but need first register there ;)

fpistm, 2020-02-21:
> @fpistm but need first register there ;)

Yes but this is the primary support. Open a new issue or commenting a new one is not a good way to get help.
Currently we try to answer/help but in the medium term, if there is too much support requested in this way, this would result to lock all comments on closed issues and reject all new one. I really don't want to get there. 😉