stm32duino/Arduino_Core_STM32 #946: ARDUINO_ARCH macro for determining platform

MCUdude, 2020-02-20:
Hi!

I'm working on a library where I need to detect what hardware is used at compile time. This is pretty straight forward for hardware such as AVR and ESP8266/ESP32. I can simply use `ARDUINO_ARCH_AVR`, `ARDUINO_ARCH_ESP8266` and `ARDUINO_ARCH_ESP32` To determine this.

I've tried to figure out where the ARDUINO_ARCH macro is defined in this core.
I found `-DARDUINO_ARCH_{build.arch}` [in the platform.txt file](https://github.com/stm32duino/Arduino_Core_STM32/blob/716f27e57fcf7649edd1c5cf23e4e0e8a6fa2230/platform.txt#L88), but I cannot figure out where build.arch comes from.

Preferably, we should have macros such as `ARDUINO_ARCH_STM32F0`, `ARDUINO_ARCH_STM32F1` and so on.

fpistm, 2020-02-20:
`ARDUINO_ARCH_STM32` identify the core.

To identify series then use `STM32YYxx`, example:
`STM32F1xx`

`ARDUINO_ARCH_STM32F1` identify Roger's core, this is the way Arduino works.

MCUdude, 2020-02-20:
Great! I will use the forum for these kinds of questions in the future.

But while we're at it;  where are these macros defined? And where does build.arch come from?

closed 2020-02-20 by MCUdude

fpistm, 2020-02-20:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification