stm32duino/Arduino_Core_STM32 #947: Pin Conflict

senceryazici, 2020-02-21:
Hi, on my Nucleo  F429ZI board, I want to use HardwareSerial1 which is connected to spesific pins in PCB, which are `PA_10, PB_6` Rx Tx respectively. I know that the PA10 is a conflict, OTG_FS... etc, but since I don't require OTG USB connector, I should be able to use it right ? How can I do such thing, thank you very much 

Ive tried
```cpp
HardwareSerial mySerial((PinName)PA_10, (PinName)PB_6)
```
but didnt work

`note: Im using pins with underscore, ex:PA_10 because PA10 is not defined due to conflict`

**Desktop (please complete the following information):**
 - OS: Ubuntu 18.04
 - Using PlatformIO
 - Upload method: STLink

**Board**
 - Name: Nucleo F429ZI

fpistm, 2020-02-21:
Hi @senceryazici 
This is normal:
https://github.com/stm32duino/Arduino_Core_STM32/blob/716f27e57fcf7649edd1c5cf23e4e0e8a6fa2230/variants/NUCLEO_F429ZI/PeripheralPins.c#L237

`PA10` is reserved for USB ID as SB125 is closed by default .
Note also, `PA10` is only available on CN12 ST Morpho  but not on CN9 as SB126 is opened by default.

Anyway, you can redefine the `const PinMap PinMap_UART_RX[] ` at sketch level as it is a WEAK and then uncomment the line for `PA_10`.

I've test it with below sketch, As I would use the 'Serial' generic name, I've selected in **U(S)ART support** menu -> **Enabled (no generic Serial)*):
```ino
HardwareSerial Serial(PA_10, PB_6);

const PinMap PinMap_UART_RX[] = {
  {PA_10, USART1,  STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART1)}, // - USB ID
  {NC,    NP,    0}
};

// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
}

// the loop routine runs over and over again forever:
void loop() {
  // read the input on analog pin 0:
  int sensorValue = analogRead(A0);
  // print out the value you read:
  Serial.println(sensorValue);
  delay(1);        // delay in between reads for stability
}
```

One other way is to use half-duplex:
https://github.com/stm32duino/wiki/wiki/API#enable-half-duplex-mode

Then in that case you do not need `PA10` (mainl if you use only Tx):
```ino
#define HALFDUPLEX

#ifdef HALFDUPLEX
HardwareSerial Serial(PB6);
#else
HardwareSerial Serial(PA_10, PB_6);
const PinMap PinMap_UART_RX[] = {
  {PA_10, USART1,  STM_PIN_DATA(STM_MODE_AF_PP, GPIO_PULLUP, GPIO_AF7_USART1)}, // - USB ID
  {NC,    NP,    0}
};
#endif

// the setup routine runs once when you press reset:
void setup() {
  // initialize serial communication at 9600 bits per second:
  Serial.begin(9600);
}

// the loop routine runs over and over again forever:
void loop() {
  // read the input on analog pin 0:
  int sensorValue = analogRead(A0);
  // print out the value you read:
  Serial.println(sensorValue);
  delay(1);        // delay in between reads for stability
}
```

senceryazici, 2020-02-21:
Oh I see thank you very much @fpistm 

One last thing,
As I understand PD8 and PD9 pins are used to connect to ST link and used with default “Serial” through USART3 (defined in ‘variant.h’)
I wonder if there is any other way to use USART3 on different pins, but also don’t loose the ST link connection. 
I know that through USART3 it is impossible but can I connect pd8 and PD9 as softwareserial instance, and define default Serial as that software serial?

Thank you very much,

Best

fpistm, 2020-02-21:
Welcome.
For those kind of question please use the forum which is the primary support way.

PD8 and 9 are physically connected to the the STLINK (which is also  STM32), this is simply a bridge.

> I wonder if there is any other way to use USART3 on different pins, but also don’t loose the ST link connection

If you mean able to use STLink to flash then yes but you will not access anymore to the VCP mount by the STLINK.

closed 2020-02-21 by fpistm